FRC-Utilities/QDriverStation #1: Checking if robot code exists...

alex-spataru, 2015-09-21:
For the moment, we managed to get almost everything working. Just today we tested the Autonomous and everything worked smoothly. For the TeleOp part, we must see how to put the joystick data in the packets sent to the robot, Wireshark will help us pave the way this week...

However, the major issue that we are facing right now is to detect if the robot code is present or not. Does any one have any idea how the DS detects it (Wireshark detected some TCP going on between the robot and the client, but the downloaded data was empty...)

ThomasJClark, 2015-09-21:
Disclaimer: I have no idea if this is 100% up to date, or if it even used to be true.  I haven't tested any of it.

Looking [here](https://github.com/WinT-3794/QDriverStation/blob/da49ddf4394ef5d9477e9644239e4df299680dab/lib/LibDS/src/Receiver.cpp#L32), it seems like QDriverStation only reads the packet index and battery level in the robot status packets, and ignores everything else.  This is my understanding of what the full packet looks like. 

    struct RobotToDSPacket {
        uint16_t packet_index;
        uint8_t  comm_version;
        uint8_t  status_byte;  // See STATUS_*
        uint8_t  program_byte; // See PROGRAM_*
        uint16_t battery;      // You already figured this one out
        uint8_t  request_byte; // See REQUEST_*
    };

    #define STATUS_TESTMODE  0x01
    #define STATUS_AUTO      0x02
    #define STATUS_ENABLED   0x04
    #define STATUS_PROGSTART 0x08
    #define STATUS_ESTOP     0x80

    #define PROGRAM_DISABLED 0x01
    #define PROGRAM_TELEOP   0x02
    #define PROGRAM_AUTO     0x04
    #define PROGRAM_TEST     0x08
    #define PROGRAM_ROBORIO  0x10
    #define PROGRAM_USERCODE 0x20

    #define REQUEST_DATETIME 0x01

You should be able to detect the presence of robot code by looking at `(packet.program_byte & PROGRAM_USERCODE) != 0x00`.

alex-spataru, 2015-09-22:
Thank you very much for the feedback, it is greatly appreciated! I will use this data to fix some issues around the library and (hopefully) detect the user code.

I will notify you tomorrow if we manage to detect the user code on the roboRIO.

alex-spataru, 2015-09-22:
We have tested the new changes and the QDriverStation is able to detect if the user code is loaded in the robot.

Thank you very much for the information!

closed 2015-09-22 by alex-spataru