FRC-Utilities/QDriverStation #2: SDL joystick mappings are different from the official FRC DriverStation

alex-spataru, 2015-09-22:
While implementing and testing the protocol, an theoretical problem has arisen: SDL generates different joystick mapping than the ones used by the official FRC DriverStation (which are provided with Xinput). Even more, this mapping can change depending on your platform and driver version.

This can pose a serious security threat for teams that do not compare the joystick input between the QDriverStation and the FRC DriverStation, because the program could cause unexpected and 'involuntary' robot movement. We could ignore this 'issue', but I would prefer to solve it....breaking a window with a robot is not a nice experience...

Some solutions that came to my mind are:
- Use a JSON file to process the data reported by SDL. This could lead to using a different file for each joystick, operating system and OS/driver version.
- Offering an option to re-map your joystick (and possibly upload them and share it with the community)
- Using the Gamepad API and assign static variables for each button and axis type (e.g A = 0, B = 1, etc)

Any ideas?

weaversam8, 2016-01-21:
+1 :+1: We also have this issue, and it has definitely been something that would be more helpful. As it is right now, one of our linux drivers is thinking about making a change in our robot code to accommodate this... I'd rather not have to do that.

alex-spataru, 2016-01-22:
Thanks! I will begin working on this issue in the next couple of days.

alex-spataru, 2016-06-29:
I will begin testing [QGamepad](https://doc-snapshots.qt.io/qt5-5.7/qgamepad.html), this can hopefully provide a good solution to the joystick mappings and help us remove the SDL dependency.

alex-spataru, 2016-11-20:
Closing this issue to avoid duplication with issue #40.

closed 2016-11-20 by alex-spataru