FRC-Utilities/QDriverStation #6: Can't compile on Ubuntu 14.04 -- How do I get qml?

PeterMitrano, 2016-01-29:
I'm unfamiliar with qmake, but running it in the root of this projects tells me i'm missing qml.
I've tried several packages (qml qtdeclarative5-dev libqt5declarative5)

What package do I need?

BearzRobotics, 2016-01-29:
This is the command I used to compile.

sudo apt-get install libsdl2-dev qtcreator qtmultimedia5-dev
libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5 && git
clone https:// <https://github.com/WinT-3794/QDriverStation.git>github.com
<https://github.com/WinT-3794/QDriverStation.git>/
<https://github.com/WinT-3794/QDriverStation.git>WinT
<https://github.com/WinT-3794/QDriverStation.git>-3794/
<https://github.com/WinT-3794/QDriverStation.git>QDriverStation.git
<https://github.com/WinT-3794/QDriverStation.git> && cd QDriverStation &&
qmake && make && sudo make install

PeterMitrano, 2016-01-29:
...can I do it without installing qtcreator? That seems like a terrible dependency.

BearzRobotics, 2016-01-29:
Yes, I had a few dependence issues with my laptop, with ubuntu  ask already
installed.

BearzRobotics, 2016-01-29:
Ubuntu sdk

PeterMitrano, 2016-01-29:
I still says I don't have qml. I'm installed all of those things except qtcreator

BearzRobotics, 2016-01-29:
Could you post the exact terminal output, so that i and other don't just
start shooting in the dark please.

One thing you could try is going to qt website and install that. You won't
have to run it as root. By not doing this it will want to install to
'/home/(UserName)'/Qt. Then its all kept separate from your main system.

http://www.qt.io/download-open-source/

i:a man dakota james owen of the family keeler am an ambassador for Christ

alex-spataru, 2016-01-29:
You can check the [travis.yml](https://github.com/WinT-3794/QDriverStation/blob/master/.travis.yml) file, as far as I know it specifies all the dependencies of the Q

PeterMitrano, 2016-01-29:
Alex you are a champ--that travis file helped a lot. Turns out qt=qt5 was required...
Thanks guys. Looking forward to testing this with FRCSim (I'm the lead on that project)

closed 2016-01-29 by PeterMitrano

alex-spataru, 2016-01-29:
I am glad that this could help you! Please let me know if I need to modify something in the QDriverStation to make it work with the FRCSim.