FRC-Utilities/QDriverStation #10: QDriverStation crashes after being compiled(Linux).

BearzRobotics, 2016-02-07:
This is running on 64 bit Ubuntu
Terminal OutPut:

QNativeSocketEngine::joinMulticastGroup() was not called in QAbstractSocket::BoundState
ASSERT: "uint(i) < uint(size())" in file /usr/include/qt5/QtCore/qbytearray.h, line 414
Aborted (core dumped)

alex-spataru, 2016-02-07:
Hello, just to know what caused the crash:

- Where you connected to the robot when the QDS crashed?
- Does it happen every time you use the QDriverStation?
- What version of Ubuntu are you running (e.g. 14.04)?
- Are you using the latest commit, or a previous version?

Sorry for asking so many questions, I should really implement a logger to trace this kind of issues for the next release!

Greetings!

BearzRobotics, 2016-02-07:
I just compiled the latest version from gitHub and was testing to see if it
ran. When it opened and then crashed , i went to a terminal and ran the app
from there and it gave the error. I'm on Ubuntu 14.04.3 LTS 64 bit install.

i:a man dakota james owen of the family keeler am an ambassador for Christ

alex-spataru, 2016-02-07:
Thanks for the information, I will try to locate the issue with a debugger and implement a simple logger so that we can check what is going wrong. I will keep you updated through this conversation.

Please note that I may be slow to respond the following days because we are currently on exams period.

BearzRobotics, 2016-02-09:
I got this error message form one of the horrible Ubuntu system problems
box.

i:a man dakota james owen of the family keeler am an ambassador for Christ

alex-spataru, 2016-02-13:
With the latest commit, I have implemented a simple logger. 
Please note that while most of the DS library is logged, I still need to log some parts of the UI client....

alex-spataru, 2016-02-28:
Hello, 

Can you try to open the QDriverStation without any external device attached to your computer? In some cases, if I have both the virtual joystick enabled and an external device attached (such as a mouse) my local installation of the QDS crashes during start-up OR when quitting. 

I am still checking what could be the causes of the crash...the debug build does not tell me anything specific for the moment.

Thanks!

BearzRobotics, 2016-02-29:
Alex it only crashed on me under my new desktop that i built. Witch the
power supply burnt up on. The Ubuntu system error box, showed that it
couldn't access something in ram. I can try to duplicate the problem on my
laptop tomorrow after school, and hope to get the same results.

i:a man dakota james owen of the family keeler am an ambassador for Christ

alex-spataru, 2016-02-29:
Thanks for the help,
Just as an additional sidenote, my installation crashed while executing [this function](https://github.com/WinT-3794/QDriverStation/blob/master/src/Joysticks/JoystickManager.cpp#L159) only when plugging in my external mouse (which for some crazy reasons reports 33 axes and 75 buttons) and enabling the virtual joystick, I am still looking if I can reproduce the crash with different conditions.

BearzRobotics, 2016-03-01:
If you install QDriverStation from a terminal run this command
 echo " " >> ~/qdriverstation.log && date >> ~/qdriverstation.log &&
~/qdriverstation >> ~/qdriverstation.log 2>> ~/qdriverstation.log

it creates and empty line at the to of a file that if it exist it will
append. It puts that data right after, and all console output from the
driver station is captured in this file for debugging.

i:a man dakota james owen of the family keeler am an ambassador for Christ

BearzRobotics, 2016-03-02:
I built the QDriverStation using the same version of qt as you, and I'm using standard sdl2 from the repository (Ubuntu 14.04). The first few things that come to my mind is a bug in  sdl2 that's in the 16.04 repo, kernel bug, or how the mouse it self interacts with the system.  

I tested this out LOGITECH DUAL ACTION controller (P /N: 863247-0010)

alex-spataru, 2016-03-02:
Well, I turns out that its a bug long-present in the Linux kernel, some keyboards and mouses are detected as a joystick. [this repository](https://github.com/denilsonsa/udev-joystick-blacklist) contains more information.

However, I still find it strange that this causes a crash on the QDriverStation, since the crash seems to be originated from the "Qt part" of the application, not from SDL or another source.

alex-spataru, 2016-06-01:
Quick & unrelated update regarding this issue: the application saves its logs on `~/.qdriverstation/`. I have also improved substantially all the code that deals with pointers and lists (which could lead to your crash issue).

If possible, could you try to reproduce your crash again and upload the logs?

BearzRobotics, 2016-06-02:
I tried but couldn't reproduce the crash on my Desktop nor on my Laptop.

Sincerely:
i:a man Dakota James Owen of the family Keeler am an ambassador for Christ

alex-spataru, 2016-06-02:
@BearzRobotics Thanks for the information! I will close this issue. If the crash occurs again, feel free to re-open this issue.

closed 2016-06-02 by alex-spataru