FRC-Utilities/QDriverStation #9: No installers for Linux

alex-spataru, 2016-02-07:
While this is not a bug. I consider this to be an issue. Linux users should be able to use the QDriverStation without needing to compile it. I would really like to support everyone, however, I am not familiar with the packaging process of _deb_ nor _rpm_ files. Note that ArchLinux is already supported by the community.

**Possible solutions:**

- If you have an idea about GNU/Linux packaging (be it Debian or Fedora distros), consider contributing to this project :pray: 
- Figure out how to use the Qt Installer Framework and build an installer with it. This would produce an installer similar to the official Qt installer. There are significant downsides to this approach: only Qt uses it, and it does not install Linux applications the way they should be installed (AFAIK).

Any help is welcome!

BearzRobotics, 2016-02-07:
I was able to make a *.run file that works here is the command, and a zip
that has the qdriverstation.run and the files i put inside of the
QDriverStation folder. They consist of a script and desktop file. Other
than that i include links to what i used to make the qdriverstation.run and
a youtube-video to learn how to use it. Best regards, let me know of any
issues that come up.

Once the qdriverstation.run is download all you have to do in a terminal is
type:
$ sudo ./qdriverstation.run

Command to make the File:
makeself.sh --bzip2 QDriverStation/ qdriverstation.run "Installer for
QDriverStation an Open source FRC Driver station" ./deploy.sh

Link to zip file, with installer and directory used to make the installer.
https://www.dropbox.com/s/xg6obdziyuqihzb/installer.zip?dl=0

MakeSelf Program
https://github.com/megastep/makeself

Video that thought me how to use it.
https://www.youtube.com/watch?v=-MLOUwoLzY8

i:a man dakota james owen of the family keeler am an ambassador for Christ

BearzRobotics, 2016-02-07:
I forgot to mention you still have to install sdl2 separate from this.

i:a man dakota james owen of the family keeler am an ambassador for Christ

alex-spataru, 2016-02-07:
Thanks for the information! I will check how this method works during the following days and keep you updated.

Greetings!

lopsided98, 2016-02-08:
Personally, I hate installing software outside of my package manager, so I would recommend making deb and rpm packages.

My experience with the deb packages is that they are fairly complicated (at least compared to Arch Linux).

BearzRobotics, 2016-02-08:
I agree, but first i wanted to start simple and work up, i'm trying to
create a ppa, but i can't get any code to upload to launch pad.

i:a man dakota james owen of the family keeler am an ambassador for Christ

BearzRobotics, 2016-02-08:
Here a while back i made an deb for 0.14, but getting it into the repose is
a challenge

https://drive.google.com/file/d/0BwmIj7Fz03lXZ1JjYnhLVVdRR0E/view

i:a man dakota james owen of the family keeler am an ambassador for Christ

viggy96, 2016-05-18:
How do I launch qdriverstation after I install the deb package? The package fails to create a desktop menu entry, and I don't know what the terminal command is to  launch it.

alex-spataru, 2016-05-18:
**TL;DR:** Try running the following command on your terminal: `qdriverstation`

**Long explanation:**

The build configuration generates a makefile that should install the QDriverStation as a normal GNU/Linux application (executable in _/usr/bin_, launcher in _/usr/share/applications_ and pixmap in _/usr/share/pixmaps_).

alex-spataru, 2016-05-19:
@viggy96 I've checked the Linux deploy configuration and it works (I had to do some fixes, because I messed up the configuration when re-writing the application in QML). However, if you used the 0.14 *.deb installer, it should have worked correctly. 

Another note, the application icon on the Unity launcher did not appear until I logged out (and logged in again). I tested this with Ubuntu 16.04.

alex-spataru, 2016-06-18:
I've begun working on creating a snap package of the QDriverStation. However, I still have some issues to tackle during the runtime:

- QProcess cannot initialize the commands used to get CPU and Battery status
- SDL2 cannot interact with the ALSA system (which does not allow the QDS to generate sounds)
- SDL2 cannot interact with the kernel correctly (which does not allow the QDS to get joystick input)

All of these issues happen because snap packages are "sandboxed" in a similar way as how Android's permissions work. Once I find the correct way to assign the proper permissions for SDL and QProcess to work, I will report back here and release the generated package for further testing.

BearzRobotics, 2016-06-18:
When you get one made I can test it on my laptop under Ubuntu and Arch
Linux.

Sincerely:
i:a man Dakota James Owen of the family Keeler am an ambassador for Christ

alex-spataru, 2016-06-19:
I finished creating a Debian package (you can get it [here](https://github.com/WinT-3794/QDriverStation/releases/download/v16.06.1/qdriverstation-16.06.1-amd64.deb)). Because I am lazy, I wrote a script that automates the process of generating debian and snap packages. You can find the scripts [here](https://github.com/WinT-3794/QDriverStation/tree/master/etc/deploy/linux), if anyone sees something that needs to be changed, please notify me :+1:

alex-spataru, 2016-06-23:
Yet another update on the snaps: If I set the snap confinement to "devmode", SDL seems to work without any issues. However, the QDriverStation still has problems getting CPU and Battery status information.

Once I fix that (maybe by reading `/proc/stat` directly instead of using CLI hacks), I will upload the obtained snap package. Please take into account that I might need to compress it, because it is very large (~45 MB), and uploading it would take forever (at least with my connection speeds).

alex-spataru, 2016-06-25:
I will close this issue and open other issues to address the following:
- Snap packages
- Setting up a PPA (AFAIK, @BearzRobotics has already created one)
- Flatpack packages (in the case that GNOME/Fedora does not allow snap packages)

Thanks for all your support!

closed 2016-06-25 by alex-spataru