FRC-Utilities/QDriverStation #11: QDriverStation crashes when quitting on OS X

nathan815, 2016-02-08:
When I quit QDriverStation, it closes but then a crash report pops up. I'm on OS X 10.9.5.

Here's the details:

```
Process:         QDriverStation [22947]
Path:            /Applications/QDriverStation.app/Contents/MacOS/QDriverStation
Identifier:      org.wint.qdriverstation
Version:         0.14 (???)
Code Type:       X86-64 (Native)
Parent Process:  launchd [159]
Responsible:     QDriverStation [22947]
User ID:         501

Date/Time:       2016-02-08 17:59:14.809 -0500
OS Version:      Mac OS X 10.9.5 (13F1603)
Report Version:  11
Anonymous UUID:  5C9BC5D4-F769-366C-3A1E-33A7C30D8CDF

Sleep/Wake UUID: 4627D1D2-D082-4004-B4C1-41F42D1F7DA2

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: EXC_I386_GPFLT

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtWidgets      	0x000000010165520b 0x10160c000 + 299531
1   org.qt-project.QtWidgets      	0x0000000101658c9a QGridLayout::~QGridLayout() + 42
2   org.qt-project.QtWidgets      	0x000000010166b3e1 QWidget::~QWidget() + 817
3   org.qt-project.QtWidgets      	0x000000010166bf8e QWidget::~QWidget() + 14
4   org.qt-project.QtCore         	0x00000001023d84d5 QObjectPrivate::deleteChildren() + 245
5   org.qt-project.QtWidgets      	0x000000010166b661 QWidget::~QWidget() + 1457
6   org.wint.qdriverstation       	0x0000000101440b3e 0x1013e5000 + 375614
7   org.qt-project.QtCore         	0x00000001023d84d5 QObjectPrivate::deleteChildren() + 245
8   org.qt-project.QtWidgets      	0x000000010166b661 QWidget::~QWidget() + 1457
9   org.qt-project.QtWidgets      	0x00000001017cd25e QStackedWidget::~QStackedWidget() + 14
10  org.qt-project.QtCore         	0x00000001023d84d5 QObjectPrivate::deleteChildren() + 245
11  org.qt-project.QtWidgets      	0x000000010166b661 QWidget::~QWidget() + 1457
12  org.wint.qdriverstation       	0x000000010144086e 0x1013e5000 + 374894
13  org.qt-project.QtCore         	0x00000001023d84d5 QObjectPrivate::deleteChildren() + 245
14  org.qt-project.QtWidgets      	0x000000010166b661 QWidget::~QWidget() + 1457
15  org.qt-project.QtWidgets      	0x000000010166bf8e QWidget::~QWidget() + 14
16  org.qt-project.QtCore         	0x00000001023d84d5 QObjectPrivate::deleteChildren() + 245
17  org.qt-project.QtWidgets      	0x000000010166b661 QWidget::~QWidget() + 1457
18  org.wint.qdriverstation       	0x0000000101428f2f MainWindow::~MainWindow() + 479
19  org.wint.qdriverstation       	0x0000000101434064 main + 1540
20  org.wint.qdriverstation       	0x00000001013ec8f4 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff88d93e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8d532f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8d535fb9 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff88d94662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff911e1421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff911e1136 _dispatch_mgr_thread + 52

Thread 3:: Qt bearer thread
0   libsystem_kernel.dylib        	0x00007fff88d939aa __select + 10
1   org.qt-project.QtCore         	0x00000001023fe584 qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) + 596
2   org.qt-project.QtCore         	0x00000001023ff450 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 672
3   org.qt-project.QtCore         	0x000000010240051a QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 234
4   org.qt-project.QtCore         	0x00000001023ad07d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
5   org.qt-project.QtCore         	0x0000000102201245 QThread::exec() + 117
6   org.qt-project.QtCore         	0x0000000102204d93 0x1021cd000 + 228755
7   libsystem_pthread.dylib       	0x00007fff8d531899 _pthread_body + 138
8   libsystem_pthread.dylib       	0x00007fff8d53172a _pthread_start + 137
9   libsystem_pthread.dylib       	0x00007fff8d535fc9 thread_start + 13
[... 331 lines cut ...]
CG backing stores                  3556K
CG image                              4K
CG raster data                       24K
CG shared images                    204K
Kernel Alloc Once                     8K
MALLOC                             43.1M
MALLOC (admin)                       32K
Memory Tag 242                       12K
Memory Tag 249                      156K
Memory Tag 251                       16K
STACK GUARD                        56.1M
Stack                              13.7M
VM_ALLOCATE                        16.4M
VM_ALLOCATE (reserved)               40K        reserved VM address space (unallocated)
__DATA                             20.5M
__IMAGE                             528K
__LINKEDIT                         70.5M
__TEXT                            117.0M
__UNICODE                           544K
mapped file                       167.2M
shared memory                         4K
===========                      =======
TOTAL                             541.3M
TOTAL, minus reserved VM space    541.2M
```

alex-spataru, 2016-02-09:
Thanks for the information! I will try to fix this issue as soon as I can. However, this may take some days, as we are currently in exam periods.

alex-spataru, 2016-02-13:
With the latest commits, the QDriverStation seems to close without any issues on Mac OS X. 
I will close this issue with the 0.15 release.

nathan815, 2016-02-14:
Awesome, thanks for fixing it!

alex-spataru, 2016-06-04:
@nathan815 Here is a download link to the release candidate: [qdriverstation-16.05-rc1](https://github.com/WinT-3794/QDriverStation/files/298626/qdriverstation-16.05-rc1.zip). 

Sorry for the (very) long delay, on one hand, there was build season & school stuff, on the other, I changed a lot of things for the new release. I hope that this release works well for you!

alex-spataru, 2016-06-12:
@nathan815 I will close this issue for now, since the latest release does not seem to crash (at least on the machines that I tested with). If the latest release still crashes with you, feel free to open this issue again. Thanks!

closed 2016-06-12 by alex-spataru