FRC-Utilities/QDriverStation #13: [Linux] SDL detects external mouse as joystick

alex-spataru, 2016-02-24:
While testing the QDriverStation on Linux (Ubuntu 14.04), I encountered a rather strange issue: SDL detects my external mouse as a joystick. Luckily, mouse movements did not affect at all the axes/buttons displayed in the Joysticks tab.

My idea to solve this issue is to create an option to 'blacklist' certain devices (using their UUID) and possibly, be able to upload the blacklist online, so that other QDriverStation installations use the online list as a reference.

Has anyone else encountered this issue? If so, what are your proposed solutions?

BearzRobotics, 2016-02-24:
This is odd. I'v never encountered this before.

alex-spataru, 2016-02-25:
Attaching screenshot for further reference. 
[image]

alex-spataru, 2016-04-10:
I've added an option to blacklist/whitelist the selected joystick in the latest commit. Its the best solution that I can come up with for the moment.

closed 2016-04-10 by alex-spataru