FRC-Utilities/QDriverStation #15: QDriverStation reports back double the battery reading

BearzRobotics, 2016-03-03:
When the QDriverStaiton reports back the battery voltage its doubled
instead of 12v it reports back as 24v. 

Using the latest build from git hub, Ubuntu 14.04 LTS 64 Bit.

Screen Shot and Log Files:
https://www.dropbox.com/sh/u8a9om7ow2tp6bp/AABJd5PbG2VSSlz5XKsZZXbwa?dl=0

alex-spataru, 2016-03-03:
I will check again the code to obtain and process the battery voltage. Thanks for reporting this issue.

alex-spataru, 2016-03-05:
I checked the the improved code today. As for me, the QDriverStation reported the correct voltage. Can you check that the issue is fixed for your team as well?

BearzRobotics, 2016-03-11:
Sorry it took so long. It turns out there is a bug in the PDP. That returns
a fake voltage on the battery. Last night in the pits we were give a new
PDP. This one does not have this problem.

alex-spataru, 2016-03-13:
I am glad that the problem was solved for you. I hope you had a great regional!

closed 2016-03-13 by alex-spataru