FRC-Utilities/QDriverStation #16: DS Hangs when docked on multi monitor setup [OS X]

makahn, 2016-03-07:
Driver station hangs when window mode is set to docked on a multi-monitor OS X setup.  

Can be avoided by leaving the window mode set to "Display Normally"

alex-spataru, 2016-03-08:
Thanks for the report! 

What version of the QDriverStation are you running and which version of Mac OS are you using?

younters, 2016-03-15:
Somewhat related, on OS X El Capitan, when the system dock is moved to the left side, the window by default is underneath the dock.

alex-spataru, 2016-03-15:
@younters Thanks! I will change the functions used to calculate the width of the window when docked, hopefully that could help to solve this issue.

alex-spataru, 2016-03-15:
@younters I fixed your issue with the latest commit, see the attached screenshot for further reference.
[image]

@makahn I already know what causes the DS to hang in a multi-monitor setup. I will check if Qt allows me to get information about the current screen in which the application is displayed, if so, your issue should be easier for me to fix.

makahn, 2016-03-18:
OS X 10.10.5
QDriverStation v0.14
- Sorry for the delay

alex-spataru, 2016-03-19:
No worries, thanks for the information!

alex-spataru, 2016-06-12:
@makahn With the latest version ([link](https://github.com/WinT-3794/QDriverStation/releases/tag/v16.06-beta)), does the QDriverStation still crash with your multi-monitor setup?

alex-spataru, 2016-12-30:
The UI code of the QDriverStation was overhauled since this issue was opened. @makahn If you experience problems with the latest version of the QDriverStation, feel free to open another issue!

closed 2016-12-30 by alex-spataru