FRC-Utilities/QDriverStation #17: [Idea] Re-write QDriverStation and LibDS with Python

alex-spataru, 2016-03-16:
Well, while C++ has worked pretty well for me and my team (and this project), most FRC programmers are familiar with Java and Python. If this project were to be maintained by the community, the most logical decision would be to adapt it to most FRC developers.

However, Java is still relatively complicated and could present some issues regarding the UI (Qt is a very good toolkit IMO). A possible solution to this could be to use Python with PyQt, which provides Python bindings to the Qt framework (which would make the whole porting process a lot simpler).

**Benefits of using Python**
- Code would be simpler to read, modify and maintain
- There are Qt and SDL bindings available for Python
- PyQt comes with a "deploy" feature for its applications, this would provide an *.exe (for Windows) and an application bundle for OS X.
- The porting process would not be very difficult, several inconsistencies in the LibDS could be fixed by doing this. 
- More potential contributors
- Better integration with GNU/Linux distros

**Cons of using Python**
- Slower start-up times and application performance, however, I want to believe that the Python byte-code compiler would take care of making the application fast enough for it to be able to communicate effectively with the robot without "losing" packets.
-  Windows users may need to install Python. It is possible that the deployment process of PyQt bundles Python with the application, so this should not be a worry.

Please share your ideas and/or suggestions!

BearzRobotics, 2016-03-16:
This is just my opinion. I would leave LIBDS in c++. While the driver
station I see no problems in bringing up to Qtpy, python or whatever.

riftware, 2016-03-16:
I agree.  If you aren't willing to do java please stay in c++      I think
a big part of the issue is that not many people know your project exists
yet.

riftware, 2016-03-16:
I just checked, there is at least one java qt bindings project.
http://qtjambi.org/

alex-spataru, 2016-03-22:
Thanks for the suggestions! I think that the QDriverStation and LibDS will continue using Qt/C++ for the moment, since I am not very familiar with Java.

@riftware Thanks for the information about QtJambi, however, QtJambi does not support Qt5 (AFAIK).

closed 2016-03-22 by alex-spataru