FRC-Utilities/QDriverStation #18: Implementing the 2014 communication protocol

alex-spataru, 2016-04-09:
I have begun working on the 2014 communication protocol (by looking at the [node-driverstation](https://github.com/gluxon/node-driverstation) code). However, I have a big problem: we have upgraded the components and wiring of our older robot (2014) to the new 2015 components as part of a training exercise for the new kids in our team during the 2015 off-season.

In other words, I cannot test if the protocol will work or not, unless I can get access to a 2013/2014 LabVIEW installation (which has been removed from NI's website), I won't be able to guarantee that my implementation will work. :confused: 

Does anyone have any papers, links, drafts, software or anything that could help me ensure that the implementation will work correctly and test it thoroughly before the release?

lopsided98, 2016-04-09:
​I can test it with our cRIO based demo robot, though I probably won't have time until after district championships next weekend.

I also have these Wireshark dissectors, which might be helpful. They cover most of the important parts of the protocol.

[frc_2014_wireshark_dissectors.zip](https://github.com/WinT-3794/QDriverStation/files/211479/frc_2014_wireshark_dissectors.zip)

alex-spataru, 2016-04-10:
@lopsided98 Thanks for the dissectors, I will take a look at them as soon as possible. I will notify you when I finish implementing the basic parts of the 2014 protocol, hopefully it will work without any major modifications.

Qormix, 2016-04-13:
I've got a working classmate laptop with the 2014 software running I could test it tomorrow

alex-spataru, 2016-04-13:
@Qormix Thanks for the support, however, I still haven't even got the base protocol right (e.g be able to switch operation modes and enable and disable the robot). In other words, the 2014 protocol will not work for the moment, and it will most probably trigger an emergency stop in your robot if you try it.

I will send a message/comment to this thread when I am relatively confident about my implementation.

alex-spataru, 2016-04-14:
Hello, here are some updates and changes that I did with the last commits:

- Generating DS-to-robot packets is almost complete, everything works (tested with @lopsided98 dissectors) except the joystick input. 
- I still need to implement the part that interprets robot-to-DS packets.
- I am not very sure if my implementation of the CRC checksum is correct.

**Questions:**
- Is it possible to know if the robot code is loaded by reading the robot-to-DS packets?
- Did the 2014 Driver Station use the POVs on the joysticks? Or it interpreted each POV as another joystick button?

Qormix, 2016-04-14:
The pov was implemented as a axis.

alex-spataru, 2016-04-15:
Another update: joystick support is now almost complete (my implementation generates axis and button data for multiple joysticks).

@Qormix Thanks for the information, I will try to collaborate your information with what I found online (it seems that nobody cares about the POVs) and the Wireshark data. Also, it seems odd that the official Driver Station (version 16.0.2, with 2014 protocol enabled) does not generate any additional data when pressing a POV...

alex-spataru, 2016-04-15:
I just implemented the basic [robot-to-DS packet interpretation code](https://github.com/WinT-3794/QDriverStation/blob/master/lib/LibDS/src/Protocols/FRC_2014.cpp#L198). For the moment, we can know if the robot has code (or not), the battery voltage and the status code of the robot (teleop, autonomous, e-stop, etc). 

It seems that robot reports if the code is present or not with the battery voltage bytes. If both of them are equal to 0x37, it means that the robot code is not present. I've got this information from [a ChiefDelphi thread](http://www.chiefdelphi.com/forums/showpost.php?p=981787&postcount=29) from 2010, but I am not very sure if it is correct.

alex-spataru, 2016-04-18:
I just have implemented and tested the CRC32 checksum and it works very well, I have also run tests with Wireshark and the QDriverStation generates the exact packet data as the official Driver Station (when both are put in the same situations). In other words, the protocol should be ready for testing on real robots.

If anyone is interested in testing the 2014 protocol, just compile the latest commit and begin testing. Just be sure to limit the area of movement of your robot (in case anything goes wrong).

I would really appreciate if anyone would give me some comments/reports on how the 2014 protocol works on their robots. I need these comments and your 'sign of approval' before releasing the new version to all users. :smiley:

alex-spataru, 2016-05-02:
I managed to obtain the 2014 Update Suite and was able to test the protocol with the LabVIEW simulator. 

I have made all necessary changes for the protocol to work as intended. Finally, I tested thoroughly my implementation to the point that I am confident that its safe for use with real robots.

@lopsided98 and @Qormix: Thanks for the support!

**Simulator w/ QDriverStation screenshot**

[image]

closed 2016-05-02 by alex-spataru

Qormix, 2016-05-02:
Works great on our robot! sorry I didn't it sooner couldn't compile it on my laptoo.

alex-spataru, 2016-05-03:
@Qormix Glad to hear it! If you find anything not working as intended, please open an issue.