FRC-Utilities/QDriverStation #19: Project Icon/Logo competition

alex-spataru, 2016-04-18:
Since this project has started, it used the logo of FIRST as its primary icon. However, this can get us in trouble with the branding policy of FIRST, since we cannot verify that every contributor and user of this project is related to FIRST. Also, the logo of FIRST can be used in many projects and does not really represent a Driver Station application....

Since I suck at designing, I would like to invite you to design and submit an icon for this project. After several submissions, the community shall vote for the best icon.

everylittlebit, 2016-04-18:
Wow, this looks like an interesting project that I will definitely be notifying my team of, here is a quick logo idea, 
[image]
Cheers from team 2702

Jelomite, 2016-04-18:
I don't know if I should stay with the FIRST colors or not, anyways, here the idea:

[image]

Good luck :smiley:

alex-spataru, 2016-04-19:
This logo was created by [Jelomite](http://github.com/jelomite).
I received an e-mail notification but somehow his/her contribution is not displayed here, so I am re-uploading the design here.

[image]

@everylittlebit Thanks for your contribution! I like that icon!

Jelomite, 2016-04-19:
@alex-spataru It seems that I accidently deleted the comment, I'm kinda new to gitHub. Thanks for reuploading!

andrewda, 2016-04-19:
@Jelomite the logo looks great!

Jelomite, 2016-04-19:
@andrewda Thank you!

alex-spataru, 2016-04-21:
@Jelomite I also like your logo, do you think that you can upload a version with a transparent background?

Jelomite, 2016-04-21:
@alex-spataru  No problem, here it is:
[image]

alex-spataru, 2016-04-22:
@Jelomite Thanks!

alex-spataru, 2016-05-03:
Based on an online [tutorial](http://gimpchat.com/viewtopic.php?f=23&t=4660), I created the this icon:

[image]

Any opinions about it?

alex-spataru, 2016-05-03:
And here is another variant of the logo with another font for the "Q"

[image]

alex-spataru, 2016-05-06:
Does anyone have more comments on the logos?

Jelomite, 2016-05-06:
@alex-spataru  The second one looks very neat, I love it.

alex-spataru, 2016-06-06:
Thanks for the contributions and comments! The QDriverStation will use the icon provided by @Jelomite.

closed 2016-06-06 by alex-spataru