FRC-Utilities/QDriverStation #20: 2014 Protocol Issues uploading code

Qormix, 2016-05-04:
QDriverStation is using a port that is required to reboot the CRio to complete an upload. this can be done manually using the reboot in QDriverstation. I will look into it further to figure out which ports and being used

alex-spataru, 2016-05-05:
Does the reboot feature not work, or is it something else that is not working properly (e.g. causing issues while uploading code)?

Qormix, 2016-05-05:
The reboot feature does work In QDriverStation works as expected. QDriverStation is preventing me from uploading code. 

The ant build is able to send code, but is unable to reboot the rio which is required to complete an upload. returning an error could not bind port already in use.

I believe this issue lies with QDriverStation having the net console running as when you compile it opens a net console link.

lopsided98, 2016-05-05:
I think I've run into this issue before, but I was using the 2016 DS using the old protocol. When trying to upload from a computer (in this case running Linux, I don't know if it happens on Windows) on the network, the build script is unable to connect to the OTA server on the cRIO, so the reboot does not happen automatically.

alex-spataru, 2016-05-05:
Thanks, I will try to change the port flags for the UDP sockets, maybe this could help fixing the issue, does anyone have any more information about how the code is uploaded to the cRIO?

lopsided98, 2016-05-05:
@Qormix I wonder if the 2016 DS connects to NetConsole when using the old protocol. That would explain both of our problem.s

lopsided98, 2016-05-05:
I know that the program is uploaded over FTP, but then the build script sends some command (maybe over NetConsole) telling the robot to reboot. The build script calls it the OTA server.

alex-spataru, 2016-05-05:
@Qormix @lopsided98 The NetConsole runs on ports 6666 and 6668 through an UDP connection. Maybe the NetConsole sockets need a special flag to be set in order for correct operation...

alex-spataru, 2016-05-05:
I found some information about the code uploading process [here](https://wpilib.screenstepslive.com/s/3120/m/7885/l/97458-understanding-the-program-download-process). As far as I know, the NetBeans plugins are open-source, if so, I could take a look at them to fix this issue.

lopsided98, 2016-05-05:
I decompiled the ant plugin, and I'm trying to figure out how it works. The important class appears to be `com.sun.spot.sdk.tool.frc.RunTask`.

lopsided98, 2016-05-05:
The OTA server appears to be separate from NetConsole, running on port 8001, and the build script just sends the string `!RUN!` over TCP to cause the cRIO to reboot.

alex-spataru, 2016-05-05:
@lopsided98 Thanks for the information! If so, I find it strange for the QDriverStation to interfere with the code uploading process. I will configure the sockets to use the SO_REUSEPORT and SO_REUSEADDRESS flags. I will report back when finished.

lopsided98, 2016-05-05:
I'm not sure if that command actually restarts the cRIO, but I can't find anything else that seems to do it. On the other hand, I can't figure out where most of the messages that appear in the output are printed, so I think I'm not looking at all the relevant code.

Next week I'm going to be spending a lot of time working on our robots, so I'll try to see if I can see anything in a live packet capture.

alex-spataru, 2016-05-06:
@lopsided98 Thanks for the help offer! I will try to find the source for the 2014 plugins, or ask at ChiefDelphi for more information regarding this topic, hopefully we can get some extra information in order to fix the issue.

Qormix, 2016-05-06:
@lopsided98 @alex-spataru I got a chance to look at the source. It appears as though it is also opening a netconsole to view a console out. which causes it to error before it can send "!RUN!" or "!DEBUG!" to the rio. Secondly the official driver station doesn't open a console stream when running in 2014 mode. That might be the best solution.

alex-spataru, 2016-05-06:
@Qormix I will disable the NetConsole for the 2014 protocol, hopefully this will be the solution.

Qormix, 2016-05-09:
Works Great! Thanks for the fix.

closed 2016-05-09 by Qormix

alex-spataru, 2016-05-10:
I am glad that the fix worked! Thanks for all the information and support!