FRC-Utilities/QDriverStation #22: QDriverStation Crashes on Connection (Mac OS X)

keco185, 2016-06-03:
Whenever the driver station software manages to show green connection it crashes within 100ms. It works up until it gets a connection to the robot however.
`Process:               QDriverStation [4243]
Path:                  /Users/USER/Downloads/QDriverStation.app/Contents/MacOS/QDriverStation
Identifier:            org.wint.qdriverstation
Version:               0.14 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QDriverStation [4243]
User ID:               501

OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000010c5c209c QCoreApplication::postEvent(QObject*, QEvent*, int) + 28
1   org.qt-project.QtNetwork      	0x000000010ca9f23e 0x10ca5c000 + 275006
2   org.qt-project.QtNetwork      	0x000000010ca95c9b 0x10ca5c000 + 236699
3   org.qt-project.QtNetwork      	0x000000010cb34ccc 0x10ca5c000 + 888012
4   org.qt-project.QtCore         	0x000000010c5f22e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
5   org.qt-project.QtNetwork      	0x000000010ca68af6 0x10ca5c000 + 51958
6   org.qt-project.QtNetwork      	0x000000010ca6a6d5 0x10ca5c000 + 59093
7   org.qt-project.QtCore         	0x000000010c5f22e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
8   org.qt-project.QtNetwork      	0x000000010ca62be4 0x10ca5c000 + 27620
9   org.qt-project.QtNetwork      	0x000000010ca64f2d 0x10ca5c000 + 36653
10  org.qt-project.QtNetwork      	0x000000010ca640de 0x10ca5c000 + 32990
11  org.qt-project.QtNetwork      	0x000000010ca6a24f 0x10ca5c000 + 57935
12  org.qt-project.QtCore         	0x000000010c5f22e2 QMetaObject::activate(QObject*, int, int, void**) + 2994
13  org.qt-project.QtNetwork      	0x000000010caf11ce 0x10ca5c000 + 610766
14  org.qt-project.QtNetwork      	0x000000010cafe97a 0x10ca5c000 + 665978
15  org.qt-project.QtWidgets      	0x000000010b84a4bb QApplicationPrivate::notify_helper(QObject*, QEvent*) + 251
16  org.qt-project.QtWidgets      	0x000000010b84d89e QApplication::notify(QObject*, QEvent*) + 8318
17  org.qt-project.QtCore         	0x000000010c5c1983 QCoreApplication::notifyInternal(QObject*, QEvent*) + 115
18  libqcocoa.dylib               	0x000000010d2c4c68 0x10d251000 + 474216
19  com.apple.CoreFoundation      	0x00007fff94d27514 __CFSocketPerformV0 + 996
20  com.apple.CoreFoundation      	0x00007fff94d08881 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
21  com.apple.CoreFoundation      	0x00007fff94ce7fbc __CFRunLoopDoSources0 + 556
22  com.apple.CoreFoundation      	0x00007fff94ce74df __CFRunLoopRun + 927
23  com.apple.CoreFoundation      	0x00007fff94ce6ed8 CFRunLoopRunSpecific + 296
24  com.apple.HIToolbox           	0x00007fff8556c935 RunCurrentEventLoopInMode + 235
25  com.apple.HIToolbox           	0x00007fff8556c76f ReceiveNextEventCommon + 432
26  com.apple.HIToolbox           	0x00007fff8556c5af _BlockUntilNextEventMatchingListInModeWithFilter + 71
27  com.apple.AppKit              	0x00007fff8b40edf6 _DPSNextEvent + 1067
28  com.apple.AppKit              	0x00007fff8b40e226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
29  com.apple.AppKit              	0x00007fff8b402d80 -[NSApplication run] + 682
30  libqcocoa.dylib               	0x000000010d27308d 0x10d251000 + 139405
31  org.qt-project.QtCore         	0x000000010c5bf07d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 381
32  org.qt-project.QtCore         	0x000000010c5c1f5a QCoreApplication::exec() + 346
33  org.wint.qdriverstation       	0x000000010b63b04a main + 1514
34  org.wint.qdriverstation       	0x000000010b5f38f4 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8a1ae5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff93eb1578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff93eaf341 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8a1aeefa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff8eccc165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff8eccbdcd _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8a1ae5e2 __workq_kernreturn + 10
[... 400 lines cut ...]
TOTAL                              1.6G      674 
TOTAL, minus reserved VM space     1.6G      674 

Model: MacBookAir5,1, BootROM MBA51.00EF.B04, 2 processors, Intel Core i7, 2 GHz, 8 GB, SMC 2.4f19
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434353153364D465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434353153364D465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.5f3 17904, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en3
Network Service: Wi-Fi, AirPort, en0
Network Service: AX88x72A, Ethernet, en7
Serial ATA Device: APPLE SSD TS128E, 121.33 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4
`

alex-spataru, 2016-06-04:
@keco185 Thanks for the report! I will release a new version (16.05) very soon. Aside from many improvements, the new version provides a much safer handling of pointers (which is a very probable cause for the crash given EXC_BAD_ACCESS).

If you want to test the new version (16.05 RC1), you can download it from here: 
[qdriverstation-16.05-rc1.zip](https://github.com/WinT-3794/QDriverStation/files/298626/qdriverstation-16.05-rc1.zip)

I would appreciate if you could try this version and notify me if it still crashes or presents any other issue on your system.

keco185, 2016-06-08:
The new versions seems to have resolved the connection problem. However, a new problem arises. It seems that the driver station can't send packets frequently enough to prevent the watchdog on the robot from triggering. Every second or so the motors cut out. There is also a graphical error where the voltage is misreported. This seems to happen when the motors cut out.
http://imgur.com/a/6ibog

alex-spataru, 2016-06-08:
@keco185 Thanks for the report! I had a very similar problem when testing out with a simulator (after I initially responded to your report). Since then, I fixed this issue (at least for me), here is the latest (hopefully fixed build): [qdriverstation-16.06.zip](https://github.com/WinT-3794/QDriverStation/files/303931/qdriverstation-16.06.zip)

I have a couple of questions:
- After loosing connection, the DS re-connected instantly? Or did you have to wait some seconds? (This to know if the problem was socket-related (as the one I fixed) or protocol-related)
- Did you use the 2014 protocol, or the newer 2015/2016 protocol? ( The 16-million volts thing bugs me... :P)

Also, thanks for the screenshots! I will fix the weird UI that is shown in your first screenshot (and check what happened with the voltage display).

keco185, 2016-06-08:
The driver station didn't disconnect in the traditional sense. The robot needs updated information from the driver station 50 times per second in order to ensure it is operating safely. For some reasons it wasn't getting updates fast enough and when the robot didn't get an update on the status of the joysticks it cut the motors for a fraction of a second until the next update came. It stayed connected and enabled, it would just cut out for a very small fraction of a second. Also, I used the 2016 protocol. Thank you. I will test the latest version at our next meeting.

alex-spataru, 2016-06-08:
Thanks! Here is (yet another!) build: [qdriverstation-16.06-threaded.zip](https://github.com/WinT-3794/QDriverStation/files/305372/qdriverstation-16.06-threaded.zip). The main difference is that this build separates some modules into different threads and (slightly) decreases the packet-sending interval to compensate for OS/hardware delays. Feel free to try this build if the other still presents problems.

keco185, 2016-06-16:
I was able to try the builds but with no success. They wouldn't connect to the robot. There was one circumstance where it flashed something about a simulation before continuing to show no connection.

alex-spataru, 2016-06-16:
@keco185 My bad, I somehow managed to make the sockets module work only for a robot with the same IP as the DS (e.g. simulating a robot on the same computer running the DS). Frankly, I do not know how or why that happened. Anyways, I re-implemented the whole sockets module again and tested it accordingly. At least for me, it worked without issues. 

You can grab a copy of the new build, by clicking [here](https://github.com/WinT-3794/QDriverStation/releases/tag/v16.06.1), or by launching the QDriverStation while your computer is connected to the Internet.

Sorry for all the buzz.

alex-spataru, 2016-06-25:
@keco185 Kevin, did the latest versions of the QDriverStation work for you?

keco185, 2016-06-25:
Sorry I completely forgot to reply. I wasn't able to get the program to connect at all.

alex-spataru, 2016-06-25:
No worries, I will close this issue (because the crash was fixed) and open a new issue regarding the QDriverStation not being able to connect to the robot. Thanks for the feedback!

closed 2016-06-25 by alex-spataru