FRC-Utilities/QDriverStation #25: Snap Packages

alex-spataru, 2016-06-25:
This is not much of an issue, it is more like a discussion channel regarding the progress of supporting snap packages.

For the moment, we are able to create and distribute snap packages, however, the generated snaps are not able to:

- Use SDL2 properly under a "strict" confinement ("devmode" works)
- Snapcraft security model does not allow applications to access `/proc/stat`, which is used by the QDriverStation to obtain CPU usage
- We use `upower` to obtain battery level & status. For the moment, snapcraft does not allow us to call `upower`.

I have already [discussed this](https://bugs.launchpad.net/snappy/+bug/1595813) with the snapcraft developers, and it is possible that these issues will be fixed in a future release of snapcraft. However, it may take a little while until snapcraft supports the interfaces / security models needed for the QDriverStation to work correctly. 

If you are interested, you can check the [snapcraft.yaml](https://github.com/WinT-3794/QDriverStation/blob/master/etc/deploy/linux/snapcraft/snapcraft.yaml) file of this project.

BearzRobotics, 2016-06-25:
Yes, I made you a co administrator when i created the team last night. The
team is seat up were anyone can ask to join, but it has to be approved by
one of the administrators.

Sincerely:
i:a man Dakota James Owen of the family Keeler am an ambassador for Christ

alex-spataru, 2016-06-25:
Thanks! For anyone that is interested to check the PPA (which can also help us with the snaps), please navigate to: https://launchpad.net/~qdriverstation

The repository is maintained by @BearzRobotics, as he said, anyone is welcome to join the team.

BearzRobotics, 2016-07-08:
Here is an update on snap packages

https://developer.ubuntu.com/en/blog/2016/07/06/announcing-new-snap-desktop-launchers/

closed 2017-02-19 by alex-spataru