FRC-Utilities/QDriverStation #26: QDriverStation not connecting to the robot (user report)

alex-spataru, 2016-06-25:
@keco185 notified me through issue #22 that his installation of the QDriverStation was not able to connect to his team's robot. The QDriverStation works for me (and hopefully for you too), but that may not be the case for everybody. 

I would like to have some feedback regarding on how the [latest (16.06.1)](https://github.com/WinT-3794/QDriverStation/releases/) version of the QDriverStation works for you. If we obtain more information regarding the environment(s) that cause the QDriverStation to fail, we can have a greater insight at what causes the connection issues and how can we fix them.

I would appreciate any kind of feedback, be it positive or negative.

Thanks in advance!

DanielEliad, 2016-07-30:
Same issue for me but I am on version 16.07

alex-spataru, 2016-07-30:
@DanielEliad Did you try to input your robot's static IP in the settings dialog? It is possible that the mDNS responder fails to find your robot (or it finds its IPv6 address, which may or may not work with Qt).

ThePlasmaGuy, 2016-09-14:
The 2014 protocol does support the CRIO on that firmware, right?  Trying to get my Mac working with an older robot we use as an outreach robot, and I can ping the robot but it won't connect via QDS (on any protocol, and I did set the default IP to the robot's IP)

alex-spataru, 2016-09-16:
@ThePlasmaGuy The QDriverStation supports the FRC 2014 Communication Protocol. If I recall correctly, you may need to change your computer's IP address manually to be able to communicate with a non-simulated 2014 robot.

ThePlasmaGuy, 2016-09-17:
@alex-spataru yeah I tried doing that and for some reason it still wouldn't connect.  Idk, the number of problems with connection I'm having it makes me think it's probably my network card or computer and not QDS.  Either that, or I just haven't figured out the trick to QDS yet lol 😂

alex-spataru, 2016-09-17:
Yeah, I find it quite strange that the robot does not respond to the QDS packets, could you upload your latest log file (you can find them in `~/.qdriverstation/`). Maybe this could help us see what is wrong :+1:

bybin, 2016-11-30:
So I installed QDS on my ubuntu 16.04 box, and it connects to the joysticks fine, but it won't connect to the robot, I tried running it as root and as a normal user with no luck, is there anyway I can fix this? It also doesn't seem to give me any errors.

alex-spataru, 2016-12-01:
@bybin Thanks for the feedback! Did you try to specify your robot's static IP in the settings dialog? The whole mDNS thing is broken with Qt....Hopefully the C version of LibDS might fix these kind of issues.

bybin, 2016-12-02:
@alex-spataru So is there anything I can do right now, or should i just wait for this to get fixed? The only thing called libDS was from 2002 and didn't do anything to help, is there something else I should be looking for?

alex-spataru, 2016-12-03:
@bybin You can find the code for LibDS here:
- C version (which will be used of the next version of QDS): https://github.com/FRC-Utilities/LibDS
- Qt/C++ version (used by the current version of QDS): https://github.com/FRC-Utilities/LibDS-Legacy

The only thing you can do for now is to specify the robot's static IP in the settings dialog (you can open it by clicking the little gear icon in the bottom-right corner of the main window). For example, my team's robot IP is *10.37.94.21*. You can find the IP of your robot by pinging it on the terminal or checking it in the [Webdashboard](http://wpilib.screenstepslive.com/s/4485/m/24166/l/262266-roborio-webdashboard).

alex-spataru, 2017-02-22:
@bybin If you haven't already, try the [latest version](https://github.com/frc-utilities/qdriverstation/releases/latest) to see if it fixed your issue. If not, please refer (or report problems) in issue #45.

closed 2017-02-22 by alex-spataru