FRC-Utilities/QDriverStation #29: QDriverStation Crashes After compilation

DanielEliad, 2016-07-16:
Hello,
I have followed the guides as described and I seem to have an issue. I tried to make sure I have the SDL and qt libraries up to date, the compilation process goes well but when I run the driver station it spouts out this error message.I am working on an ubuntu 14.04.

00:00.0        DEBUG         Scale factor set to: 1 
terminate called after throwing an instance of 'std::regex_error'
  what():  regex_error
Aborted (core dumped)

Is this error a familiar one?,
Thank you very much, can't wait to use this, looks great.

alex-spataru, 2016-07-16:
Well, I did not stumble with this error before. Some Google searches indicate that this error could be related to GCC (e.g. using an older version that does not fully support C++11). Could you try to follow the commands outlined in the [Travis build file](https://github.com/FRC-Utilities/QDriverStation/blob/master/.travis.yml)?

Anyways, I will install Ubuntu 14.04 on a VirtualBox and see what happens. If I find a solution, I will post it here. 

Thanks for the report!

alex-spataru, 2016-07-16:
**Update** It seems that your error comes from the code that gets CPU usage under Linux. We use [regex](http://www.cplusplus.com/reference/regex/) to do that, which only works with C++ 11 or greater. Updating your GCC to a later version (e.g what the [Travis CI file](https://github.com/FRC-Utilities/QDriverStation/blob/master/.travis.yml) does) should fix your issue.

References:
- http://www.cplusplus.com/reference/regex/
- http://stackoverflow.com/questions/8060025/is-this-c11-regex-error-me-or-the-compiler

Let me know if the above solution works for you!

GreenBlitz4590Programmers, 2016-07-17:
Thank you for answering,
I have tried both with gcc and g++ version 4.8 and with them version 4.9 that supports c++11 but didn't manage to run the program.

alex-spataru, 2016-07-17:
@GreenBlitz4590Programmers @DanielEliad I will try to replace the standard regex functionality with Qt's built-in regex functionality. Hopefully this will fix this issue.

alex-spataru, 2016-07-17:
**Update** I re-wrote the code that obtained CPU usage under GNU/Linux using Qt functions. The application should compile and run fine, regardless of GCC version or C++11 support.

@DanielEliad @GreenBlitz4590Programmers Can you try the latest commit to see if this issue was solved?

GreenBlitz4590Programmers, 2016-07-17:
Thank you so much for helping, i'm trying it out right now. actually Daniel and I are the same person. This account is my FRC tean's github account.

GreenBlitz4590Programmers, 2016-07-17:
Unfortunately, it didn't work. However, now I am getting a different error message,
00:00.0        DEBUG         Initializing DriverStation... 
00:00.0        DEBUG         DriverStation initialized! 
00:00.2        WARNING       QQmlApplicationEngine failed to load component 
00:00.2        WARNING       qrc:/qml/main.qml:86 Type MainWindow unavailable
qrc:/qml/MainWindow/MainWindow.qml:27 module "Qt.labs.settings" is not installed
 
00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.5        DEBUG         DS networking operations stopped 
00:00.5        DEBUG         Log buffer closed 
00:00.5        WARNING       QObject::startTimer: Timers can only be used with threads started with QThread

Thanks for all your help,
Daniel

alex-spataru, 2016-07-17:
Well at least we got our first issue solved :+1: 

For the second error to be fixed, you must install [qml-module-qt-labs-settings](http://packages.ubuntu.com/search?keywords=qml-module-qt-labs-settings). If you are using Ubuntu 14.04 or 14.10, you may need to download and install this package manually, since it is available for Ubuntu 15.04 or greater.

GreenBlitz4590Programmers, 2016-07-19:
I have been trying to find a way to install this package manually, I found a .deb file and installed it but now I am getting this message saying the version does not suffice:

00:00.0        DEBUG         Initializing DriverStation... 
00:00.0        DEBUG         DriverStation initialized! 
00:00.2        WARNING       QQmlApplicationEngine failed to load component 
00:00.2        WARNING       qrc:/qml/main.qml:86 Type MainWindow unavailable
qrc:/qml/MainWindow/MainWindow.qml:27 plugin cannot be loaded for module "Qt.labs.settings": The plugin '/usr/lib/x86_64-linux-gnu/qt5/qml/Qt/labs/settings/libqmlsettingsplugin.so' uses incompatible Qt library. (5.3.0) [release]
 
00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.6        DEBUG         DS networking operations stopped 
00:00.6        DEBUG         Log buffer closed 
00:00.6        WARNING       QObject::startTimer: Timers can only be used with threads started with QThread

I have installed both qml-module-qt-labs-settings version: 5.3.0 and 5.4.2
I have tried installing version 5.5.1 but htis outputs the message:
Selecting previously unselected package qml-module-qt-labs-settings:amd64.
(Reading database ... 221750 files and directories currently installed.)
Preparing to unpack .../qml-module-qt-labs-settings_5.5.1-2ubuntu8-2_amd64.deb ...
Unpacking qml-module-qt-labs-settings:amd64 (5.5.1-2ubuntu8~2) ...
dpkg: dependency problems prevent configuration of qml-module-qt-labs-settings:amd64:
 qml-module-qt-labs-settings:amd64 depends on libqt5core5a (>= 5.5.0); however:
  Version of libqt5core5a:amd64 on system is 5.2.1+dfsg-1ubuntu14.3.

dpkg: error processing package qml-module-qt-labs-settings:amd64 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 qml-module-qt-labs-settings:amd64

Do you have any ideas?

Thanks in advance :+1:

alex-spataru, 2016-07-20:
The only thing that I can think of is adding a PPA to your system and upgrading Qt with it, you can do this with the following commands:

```
sudo add-apt-repository ppa:beineri/opt-qt57-trusty # Add repository
sudo apt-get update                                 # Update package list
sudo apt-get install qt57*                          # Install every Qt 5.7 component
sudo sh /opt/qt57/bin/qt57-env.sh                   # Setup correct environment (according to repo)
```

**Note:** I have not tested these commands, I am just following the repository instructions.

Source: https://launchpad.net/~beineri/+archive/ubuntu/opt-qt57-trusty

Sorry for the late reply,
--Alex

DanielEliad, 2016-07-25:
It worked!

Thank you so much for your help :smile:

alex-spataru, 2016-07-25:
@DanielEliad Glad to hear it! Thanks for the report!
@GreenBlitz4590Programmers Did you manage to fix your issues? Maybe we still can help you out.

DanielEliad, 2016-07-25:
We are the same person actually, one account is just my FRC team's account and the other is my private account.

I am having one issue which may be present in the original driver station for windows. I am using 2 Logitech Extreme joysticks and one XBOX controller. When I connect all three the driver station(both in windows and linux) detects two joysticks as the same one, so when I move either joystick the axis values change in a single joystick element. There is another joystick element, but no actual joystick changes it's values.

Did you hear about this issue before?

DanielEliad, 2016-07-25:
Also now I appear to not have connection through QDriverStation but I do have a connection to the robot through wine.

DanielEliad, 2016-07-25:
Maybe these warnings have to do with it:

00:00.6        WARNING       QSslSocket: cannot resolve SSLv2_client_method
00:00.6        WARNING       QSslSocket: cannot resolve SSLv2_server_method

alex-spataru, 2016-07-25:
These warnings are about the updater, nothing to worry about. I have received other issues regarding the communications...I am uploading the fixed builds right now :)

Regarding the joysticks, I will open a new issue about them (so that we keep the issues organized) and mention you there.

closed 2016-07-25 by alex-spataru