FRC-Utilities/QDriverStation #31: Older Protocols?

MC42, 2016-08-05:
The front-page lists QDS as supporting back to 2009 and yet the drop-down only goes back to 2014.  Were the older versions dropped, or backstashed somewhere else in the code?

alex-spataru, 2016-08-05:
AFAIK, the DS 2009 through 2014 communication protocols where the same (at least with what I found on GitHub and ChiefDelphi). Selecting the 2014 protocol should allow you to control any 2009-2014 robot. 

On the other hand,  I have separated the 2016 and 2015 protocols because the base robot address was changed between the seasons (from roboRIO-XXYY.local to roboRIO-XXYY-FRC.local), but the rest of the implementation is the same.

As a final note, my team did not keep their robots prior to 2014, since we could not afford to buy components each new year, so its possible that my assumption is incorrect.

MC42, 2016-08-05:
Next time I'm in the shop, I'll test on our 2012 robot with the cRIO.  Will post results.

alex-spataru, 2016-08-05:
Thanks, if you need help with anything, just reply to this thread.

https://github.com/FRC-Utilities/QDriverStation/issues/31#issuecomment-237752253

MC42, 2016-08-13:
Upon further inspection, the QDriverStation *does* support the protocol, and can successfully connect for Comms.  However, it flashes the "No Robot Code" error upon an attempt to enable.  This isn't needed upon using the old vanilla DS or the "FRC Drive" mobile app.

alex-spataru, 2016-08-13:
Thanks for the information! I will change the 2014 protocol implementation to fix your issue.

MC42, 2016-08-13:
Thanks so much!

alex-spataru, 2016-08-16:
I made a patch to the 2014 protocol implementation, which should fix your issue (not detecting robot code). I will draft a new release and reference it in this thread when finished.

Thanks for the report!

alex-spataru, 2016-08-16:
The fixed version is available [here](http://github.com/frc-utilities/qdriverstation/releases/latest). I hope it works nicely for you!

MC42, 2016-08-16:
Thanks so much!  I'll test it (again :P) next time I'm in the shop!

closed 2017-02-22 by alex-spataru