FRC-Utilities/QDriverStation #33: Changing Rio Address causes crashes

bman12three4, 2016-09-02:
I changed the rio address from roboRIO-5482-FRC.local  to roboRIO-5482-FRC.local:22 and the application crashed, and it keeps crashing every time I try to reopen it.

BearzRobotics, 2016-09-02:
What version are you using? Try opening it from the command prompt/Terminal
to see what is happing. Plus if your version is new a enough you have a log
file in
(Windows) C:\Users\<Name>\.qdriverstation
(Linux And probably Mac) /home/<user Name>/.qdriverstation

Sincerely:
i:a man Dakota James Owen of the family Keeler am an ambassador for Christ

alex-spataru, 2016-09-02:
The address you gave is invalid (you should not specify a port in the custom address). The QDriverStation crashes when trying to find the IP of the address that you gave.

I can help you to reset the settings of the QDriverStation, but I need to know which operating system are you using (the procedures are a bit different for each OS).

Thanks!

https://github.com/FRC-Utilities/QDriverStation/issues/33

bman12three4, 2016-09-02:
I am using Mac OS 10.10.5, and the latest version of qDriverStation

bman12three4, 2016-09-02:
[2016_09_02 15_19_43 Fri.txt](https://github.com/FRC-Utilities/QDriverStation/files/452752/2016_09_02.15_19_43.Fri.txt)
Here is the log file

alex-spataru, 2016-09-03:
Thanks for the information!

To reset the settings of the QDriverStation (which will fix the crash), open a Terminal and type the following:

> cd /Applications/QDriverStation.app/Contents/MacOS

> ./QDriverStation --reset

Sorry for the late reply, I can only access the Internet with mobile data for the moment (which is slow and expensive with Telmex).

https://github.com/FRC-Utilities/QDriverStation/issues/33#issuecomment-244464884

bman12three4, 2016-09-06:
hmm, It says it reset but is still giving me a segmentation fault whenever I try to run it.

alex-spataru, 2016-09-06:
I ran into this issue too, resetting the QDriverStation and rebooting the computer worked for me. I believe that OS X has a "settings cache" feature or something similar, which prevents a full reset of an application.

I will upload a new release of the QDriverStation which ignores faulty addresses. However, this may take a while,  since I have very limited access to the Internet right now.

https://github.com/FRC-Utilities/QDriverStation/issues/33#issuecomment-244977825

bman12three4, 2016-09-16:
I fixed this issue manually by changing the settings to report back a valid IP, then ran the program. This seemed to clear it and then I reverted the code back and the program ran stable again. I will make sure to only use valid IP's from now on!

alex-spataru, 2016-09-17:
Glad to hear that! I will improve the address validation filter as I integrate the new LibDS-C with the QDriverStation to avoid similar crashes. I will keep you notified :+1:

closed 2016-10-06 by bman12three4