FRC-Utilities/QDriverStation #35: QDS crashes upon opening sub-window after attempting roborio connection.

ThePlasmaGuy, 2016-09-07:
I installed QDriverStation last night, and all seemed to work well.  However, I didn't have an FRC Robot to test connection, so I waited until our meeting today.  After connecting to the robot's wifi and opening QDS, absolutely nothing happens.  And to make matters worse, QDS now refuses to open any context windows.  If I try to open up the QDS settings or the virtual joystick menu, QDS crashes.  I tried rebooting and switching out my .app, but nothing seems to alleviate the issue.  

My thought was that QDS is storing some files somewhere on my machine that have been corrupted.  Is this true, and if so, where would I find them on Mac OS X El Capitan v10.11.6?  

I downloaded QDS to use as a backup driver station since the normal driver station is Win-only.  I'd really like to be able to accomplish this goal without having to install the DS or QDS in a VM.  Thanks for your help :)

ThePlasmaGuy, 2016-09-14:
Totally forgot about this issue.  I was able to get it to work by closing down QDriverStation, rebooting my computer, and then waiting.  I tried resetting QDS but that didn't really help, I guess I just had to let it sit and then it started working again lol.  Once I could get the dialog to open, setting the default IP worked fine.

closed 2016-09-14 by ThePlasmaGuy

alex-spataru, 2016-09-16:
Glad it worked! Thanks for trying out the QDS!