FRC-Utilities/QDriverStation #37: module "QtQuick.Controls" is not installed

calebeby, 2016-09-12:
When I run `qdriverstation`, I get this:
```bash
using qt5ct plugin
00:00.1        DEBUG         Initializing DriverStation...
00:00.2        DEBUG         DriverStation initialized!
00:00.2        WARNING       QQmlApplicationEngine failed to load component
00:00.2        WARNING       qrc:/qml/main.qml:111 Type LogsWindow unavailable
qrc:/qml/Dialogs/LogsWindow.qml:26 module "QtQuick.Controls" is not installed

00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.2        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.2        DEBUG         DS networking operations stopped
00:00.2        DEBUG         Log buffer closed
00:00.2        WARNING       QObject::startTimer: Timers can only be used with threads started with QThread
```

I don't know how to install QtQuick.Controls.

alex-spataru, 2016-09-13:
Hello, check that you have the following packages installed in your system:

- libqt5gui5
- libqt5qml5
- libqt5quick5
- libqt5widgets5
- libqt5network5
- libqt5multimedia5
- libqt5declarative5
- qml-module-qtquick2
- qml-module-qtquick-window2
- qml-module-qtquick-layouts
- qml-module-qtquick-controls
- qml-module-qt-labs-settings

The QDriverStation is tested with Ubuntu 15.10 and above. If you are using an older system (e.g. a distro based on Ubuntu 14.04), you may need to obtain and install some of these packages manually ([this](http://packages.ubuntu.com/xenial/libs/) may help you!).

Sorry for the late reply!

calebeby, 2016-09-13:
I am running arch, and the AUR thing didn't work.

calebeby, 2016-09-13:
Now, I'm re-running the install, it looks like it's getting farther than it did yesterday.

lopsided98, 2016-09-13:
I just fixed that a couple of days ago, so it should install now.

lopsided98, 2016-09-13:
I just realized that I forgot to fix the -git package, if that's what you're running then it still won't work. I'll fix it right now.

calebeby, 2016-09-13:
Yeah, yesterday I was installing the -git package, today the qdriverstation package is getting farther.

alex-spataru, 2016-09-13:
@lopsided98 If you need the full list of dependencies (build & release), check the [Snapcraft](https://github.com/FRC-Utilities/QDriverStation/blob/master/etc/deploy/linux/snapcraft/snapcraft.yaml) build config, it contains every dependency required to use the QDriverStation.

lopsided98, 2016-09-13:
The -git package should work now.

calebeby, 2016-09-13:
The not -git package worked, I did not test the -git one. This looks really cool, except that I'm in a tiling WM (i3). What happens when the window is too big for the DS is it jumps up and down. I had to make the window a floating window to make this stop. Maybe you can make it just stick to the bottom of the window?

alex-spataru, 2016-09-13:
You have an option to "dock" the window in a manner similar to the official DS...maybe it could work for you.

I have never used i3 or tested QDS on it, so my assumption could be wrong.

calebeby, 2016-09-13:
When it is jumping up and down it is nearly impossible to click the button to dock it.

calebeby, 2016-09-13:
When I am able to click the dock button it only makes the hum sound, but does not dock. i3 overrides the window's dimensions if the window is tiled (which, by default, it is).

alex-spataru, 2016-09-13:
Maybe we can tweak the window flags to fix your issue, however, its up to the WM to respect those flags.

calebeby, 2016-09-13:
I think it's OK if it's tiled by default, as long as it the content doesn't
jump up and down.

alex-spataru, 2016-09-16:
@calebeby To keep things organized, I will create a new issue with your WM's problems and mention you there.

closed 2016-09-16 by alex-spataru