FRC-Utilities/QDriverStation #38: Problems with i3 window manager

alex-spataru, 2016-09-16:
In a previous issue, @calebeby mentioned that the QDriverStation does not behave as expected with the i3 window manager.

Has anyone else experienced issues with i3 and the QDS? If so, what have you done to fix your issue? What solutions would you suggest?

@calebeby Could you please upload your latest log of the QDS? (they are located in `~/.qdriverstation/`)

calebeby, 2016-09-16:
https://gist.github.com/calebeby/57f5b54b707323005bfe9a956b9e2c9f

alex-spataru, 2016-09-16:
Thanks!

I have found the following warnings (which could be relevant in this case):

```
00:03.1        WARNING       propsReply "Method \"GetAll\" with signature \"s\" on interface \"org.freedesktop.DBus.Properties\" doesn't exist\n"
00:03.1        WARNING       nmReply "Method \"GetDevices\" with signature \"\" on interface \"org.freedesktop.NetworkManager\" doesn't exist\n"
00:03.1        WARNING       "Object path cannot be empty"
00:03.2        WARNING       "Object path cannot be empty"
```

QML did not report anything wrong with the window geometry, maybe manually setting some window flags during app initialization could fix this issue (comments are welcome).

calebeby, 2016-09-16:
Sure, I have no idea how window flags work, but if you want me to test something I can.

calebeby, 2016-09-16:
Maybe this part is relevant?

```
00:02.5        WARNING       file:///usr/lib/qt/qml/QtQuick/Controls/Private/BasicTableView.qml:613:17: QML Item: Binding loop detected for property \"width\"
00:02.5        WARNING       file:///usr/lib/qt/qml/QtQuick/Controls/Private/BasicTableView.qml:613:17: QML Item: Binding loop detected for property \"width\"
```

alex-spataru, 2016-09-16:
Sure! I will write to this thread when I make a change to the window flags or when I get a chance to test the QDS with i3. 

In this part:
```
00:02.5        WARNING       file:///usr/lib/qt/qml/QtQuick/Controls/Private/BasicTableView.qml:613:17: QML Item: Binding loop detected for property \"width\"
00:02.5        WARNING       file:///usr/lib/qt/qml/QtQuick/Controls/Private/BasicTableView.qml:613:17: QML Item: Binding loop detected for property \"width\"
```
QML complains about some ugly hacks that I did to implement the joysticks tab. It should not affect the window size at all.

closed 2020-10-07 by alex-spataru