FRC-Utilities/QDriverStation #39: QDriverStation unable to establish connection with robotRIO

primetoxinz, 2016-10-18:
Specs:
Arch Linux 4.7.6-1
QDS Commit [9393cc2](https://github.com/FRC-Utilities/QDriverStation/commit/9393cc2115ef63a6987d6e91c6f2babf73009fee)
RoboRIO Firmware: unknown currently, likely last 2015 version

Today I tested connecting to the roboRIO with QDS for the first time.
The wireless connection was successfull and the roboRIO web dashboard (10.17.93.1) works perfectly.
Launching QDS there is an attempt to connect to roboRIO-1793-FRC.local and no other output
Attempted changing the mDNS to the static ip of the roboRIO (10.17.93.76) still no change
Tried the official FRC DS on the provider FRC laptop, immediately established comms.

This is either an issue with QDS or my networking setup, any ideas?
If any info from the roboRIO is necessary I won't have access until this wednesday

BearzRobotics, 2016-10-18:
I would suggest looking at the log file(s) in '~/.qdriverstation/'

Sincerely:
i:a man Dakota James Owen an ambassador for Christ.
* ALL RIGHTS RESERVED*

primetoxinz, 2016-10-18:
[here](http://pastebin.com/5N1cB6Jh) is the log for the first attempt to use QDS

bman12three4, 2016-10-18:
Just to piggy-back on this, this occasionally happens to me as well. The solution for me is to delete the app from my computer and reinstall it, and that works for a couple more times.

primetoxinz, 2016-10-18:
hopefully will be able to try the reinstall trick tomorrow, would still be nice to know the cause of this and possibly find a fix, if it does work.

BearzRobotics, 2016-10-19:
It looks like first a problem with the
QNetworkReplyImplPrivate. Second it looks like a hex value is trying to be
read as a function.

Third this right here looks like QObject::disconnect needs a parameter but
non was given.
"QObject::disconnect: Unexpected null parameter"

I believe this is the underlying problem.  Tomorrow  i will email Alex
directly to get his attention.
============================
\"roboRIO-1793-FRC.local\" scheduled for lookup
01:23.2        WARNING       QObject::disconnect: Unexpected null parameter
01:38.1        WARNING       QNetworkReplyImplPrivate::error: Internal
problem, this method must only be called once.
03:13.2        DEBUG         Robot reboot triggered by DS...
03:14.9        WARNING       qrc:/qml/MainWindow/Diagnostics.qml:102:
TypeError: Property 'restartCode' of object DriverStation(0x777c00) is not
a function
06:16.1        WARNING       qrc:/qml/MainWindow/Diagnostics.qml:102:
TypeError: Property 'restartCode' of object DriverStation(0x777c00) is not
a function
06:16.8        WARNING       qrc:/qml/MainWindow/Diagnostics.qml:102:
TypeError: Property 'restartCode' of object DriverStation(0x777c00) is not
a function
06

alex-spataru, 2016-10-19:
Sorry for the late response, I just got back from school right now.

There are several possible causes for your issue:

- The robot is already connected to an existing DS (QDriverStation or FRC Driver Station) instance, make sure that that the Driver Station is not opened on another computer in your LAN (or robot network)
- For some reason, the mDNS lookup(s) fail and cannot find the robot (try using your robot's static address, which is applied only when you hit the 'Close' button on the preferences window)

I think that the main problem is that your custom address was never used by the QDS (judging on the log file). Try clicking the close button on the preferences window after imputing the robot's static address. 

Hopefully this will fix your issue!

primetoxinz, 2016-10-19:
I inpu

closed 2016-10-19 by primetoxinz

alex-spataru, 2016-10-19:
@primetoxinz Glad you've got it to work. Getting mDNS to work on every OS is not so easy, even when the QDriverStation has its own fallback implementation of mDNS. Hopefully one day we'll be able to make it work seamlessly 👍