FRC-Utilities/QDriverStation #40: Joystick mappings are wrong

RikuKawai, 2016-11-19:
Joystick mappings for the Logitech F310 are incorrect in QDriverStation, but are correct in jstest-gtk.
Debian Stretch x86_64, Linux 4.7, QDriverStation 16.10.

They should be as follows, which matches what they are under Windows and in the robot code:

- Axis 0: Left Stick X
- Axis 1: Left Stick Y
- Axis 2: Left Trigger
- Axis 3: Right Stick X
- Axis 4: Right Stick Y
- Axis 5: Right Trigger

In QDriverStation, the mappings are:

- Axis 0: Left Stick X
- Axis 1: Left Stick Y
- Axis 2: Right Stick X
- Axis 3: Right Stick Y
- Axis 4: Left Trigger
- Axis 5: Right Trigger

This renders it impossible to correctly operate the robot without having different code for Windows and Linux.

alex-spataru, 2016-11-19:
Hi!

This issue has been present for a [very long](https://github.com/FRC-Utilities/QDriverStation/issues/2) time. Since we use SDL for cross-platform joystick support, we (unfortunately) depend on the mappings provided by SDL.

For the moment, I think that the smartest way is to implement a way to allow the user to re-map the joysticks (and possibly sharing the *fixed* mappings with the community). I will work on this issue when I have some free time in December, as I am currently about to present final exams for this quarter. 

Thanks for the feedback!

everylittlebit, 2016-11-20:
Can confirm, Same issue on our Logitech Attack 3's

alex-spataru, 2016-11-20:
Thanks for the confirmation! I will try alternative options to SDL, such as [Gainput](http://gainput.johanneskuhlmann.de). Ultimately, I think that the best way to standardize joystick input across operating systems will be to allow users to manually re-map joysticks on the fly...

everylittlebit, 2016-11-24:
Interestingly, the output from QJoysticks is different from the output of [sdl-jstest](https://github.com/Grumbel/sdl-jstest), but only on one axis

alex-spataru, 2016-12-01:
@everylittlebit This can happen because QJoysticks uses SDL2 instead of SDL.

edelmanjm, 2017-10-11:
Still having this issue on macOS High Sierra 10.13. I agree with @alex-spataru that the best solution seems to be to allow the user to remap the controls. Unfortunately AFAIK [360Controller](https://github.com/360Controller/360Controller) doesn't contain any way to remap axes, only buttons.