FRC-Utilities/QDriverStation #42: Unable to build on rPi

fovea1959, 2016-11-29:
got a good(?) build on x64. Trying to build on rPi/raspbian.

qmake yields:
    Project ERROR: Unknown module(s) in QT: quick qml

I have installed every package I could find that looked applicable; what does it take to build this on raspian?

BearzRobotics, 2016-11-29:
from the snapcraft
<https://github.com/FRC-Utilities/QDriverStation/blob/master/etc/deploy/linux/snapcraft/snapcraft.yaml>
file these are the dependency it needs to build in a containerized snap.
Hope it help, Thursday I can try and set up my raspberry pi 3 and try to
compile it.

- libsdl2-dev
- qtbase5-dev
- qttools5-dev
- build-essential
- libqt5webkit5-dev
- qtmultimedia5-dev
- qtdeclarative5-dev
- qttools5-dev-tools
stage-packages:
- libsdl2-2.0-0
- libqt5gui5
- libqt5qml5
- libqt5quick5
- libqt5widgets5
- libqt5network5
- libqt5multimedia5
- libqt5declarative5
- qml-module-qtquick2
- qml-module-qtquick-window2
- qml-module-qtquick-layouts
- qml-module-qtquick-controls
- qml-module-qt-labs-settings

Sincerely:
i:a man Dakota James Owen an ambassador for Christ.
* ALL RIGHTS RESERVED*

fovea1959, 2016-11-30:
Have the same problem on my home ubuntu x64 system. work ubuntu system is fine.

have installed every package from the snappackages list above on home system, problem persists....

alex-spataru, 2016-12-01:
Sorry for the late reply, I have been somewhat busy lately. Your error occurs because you do not have the QML/Quick module installed, which is used by the QDriverStation to draw the UI.

I will check if rPi has a package for QML available. If not, you can still try the Console DS, which has [minimal dependencies](https://github.com/FRC-Utilities/LibDS/tree/master/examples/ConsoleDS#dependencies) (and also supports reading joysticks).

If you are interested, download [LibDS](https://github.com/FRC-Utilities/LibDS) and navigate to the [examples/ConsoleDS](https://github.com/FRC-Utilities/LibDS/tree/master/examples/ConsoleDS) directory. Just modify [this line](https://github.com/FRC-Utilities/LibDS/blob/master/examples/ConsoleDS/src/main.c#L47) in *main.c* to point to your robot's static IP address.

fovea1959, 2016-12-01:
I am trying to track the issue down on my x64 Ubuntu also; the package must be available there because it builds fine on one of my Ubuntu machines, but not the other. Trying to figure out which package is missing.

fovea1959, 2016-12-01:
If you have multiple version of QT installed, you need to force qmake to use Qt5.

    QT_SELECT=qt5 qmake

on my offending x64 system, qt4 seemed to be the default, and I don't think qt4 had all the pieces.

Probably need to update the build instructions. Will try this on rPi tonight.

fovea1959, 2016-12-01:
Alex: how to build stuff in LibDS? Not familiar with .pro files.

alex-spataru, 2016-12-02:
Thanks for the hint! I will modify the project to force Qt5 usage by default.

fovea1959, 2016-12-02:
on rPi, needed to apt-get install qml-module-qtquick-control, I believe.

     QT_SELECT=qt5 qmake; make

worked fine.

Resultant executable won't run.

alex-spataru, 2017-02-19:
@fovea1959  Sorry for the (very) late reply. Could you try to build the [latest](https://github.com/FRC-Utilities/QDriverStation/archive/master.zip) version of the QDriverStation? Also, ensure that you have the dependencies mentioned by @BearzRobotics installed on your system.

closed 2019-01-03 by alex-spataru