FRC-Utilities/QDriverStation #45: No robot communication

ArtificialAmateur, 2017-01-12:
After connecting to the radio AP through a laptop running Linux Mint QDriverStation does not detect any robot communication, but the official version on a Windows laptop does.

alex-spataru, 2017-01-12:
It is possible that your distro does not come with a package to handle mDNS requests correctly, or that Qt has problems interacting with the mDNS equivalent on your system.

To fix your issue, try to specify your robot's static IP in the settings dialog (you can open it by clicking the little gear icon in the bottom-right corner of the main window). For example, my team's robot IP was (if I recall correctly) _10.37.94.21_. You can find the IP of your robot by pinging it on the terminal or checking it in the _WebDashboard_.

Qormix, 2017-01-12:
The default address for the Rio should be 10.XX.YY.20

 Try opening up your terminal and pinging roborio-XXYY-frc.local this will let you know if mDNS is part of you Linux distro.

ArtificialAmateur, 2017-01-13:
I have avahi installed by default. I cant find the roboRIO ip nor the webdashboard ip.

Qormix, 2017-01-14:
which version of mint do you have installed I'll see if I can reproduce this on one of my laptops

alex-spataru, 2017-01-15:
@ArtificialAmateur I will upload a new release later today, which should fix your communication issues by using native functions. Also, please ensure that you click the 'Close' button in the settings dialog to save and apply your changes. I have clarified the button labels in the new release, since this was a very common mistake by some users.

ArtificialAmateur, 2017-01-17:
Found out you have to powercycle the roboRIO while QDriverStation is open.

ArtificialAmateur, 2017-01-23:
roboRIO no longer connects with new version.

alex-spataru, 2017-01-23:
@ArtificialAmateur Could you please upload a log file?

Qormix, 2017-01-28:
I'm seem to be experiencing the same issues with Ubuntu and with the 2017 robot update. QDriverstation works fine on windows and the 2017 update. A combination of restarting the rio and qdriverstation. switching between mDNS and static will get it to connect eventually.

chackotaco, 2017-01-28:
connecting rarely works for me. I was only able to get it to connect once, and I'm not sure how...

Some odd behavior I am able to consistently reproduce:
If you open qdriverstation from the command line, and then open another instance of qdriverstation at the same time from the command line, one the the qdriverstations will successfully find the robot for a few seconds...  I could verify this by removing robotcode (the it shows it has communications but no robot code)

Qormix, 2017-01-28:
I've also found that. the console output works.

alex-spataru, 2017-01-30:
@Qormix @chackotaco Was the QDriverStation able to connect previously to the 2017 robots? I suspect that this is an issue with the socket flags under Linux or something...

chackotaco, 2017-01-30:
I was never able to get qdriverstation on 2017 robots to connect except for one instance.. which is frustrating from a debugging standpoint. The 2016 robots and 2016 qdriverstation worked consistently however.

Qormix, 2017-01-30:
I'll have a chance with a rio tomorrow I'll get you some logs then.

snapp-development, 2017-01-31:
I am also unable to communicate with our RoboRios falshed with the 2017 firmware. I am using macOS Sierra Version 10.12.2. I attached my logs if
[logs.zip](https://github.com/FRC-Utilities/QDriverStation/files/743153/logs.zip)

 that helps.

alex-spataru, 2017-02-02:
@Qormix @chackotaco @snapp-development Thanks for the logs and information! I think that this issue has something to do with how LibDS resets the sockets when a watchdog expires. I will send you some experimental builds for Ubuntu to check if that fixes the problem.

rolylolypoly, 2017-02-07:
So I'm just going to document my struggle here real quick:
If your RIO's Comm light is green, but your DS status lights are red, run QDS as sudo.
If you are also using Smartdashboard, you have to run that seperately as sudo.

alex-spataru, 2017-02-11:
@Qormix @chackotaco @rolylolypoly @snapp-development Sorry for the late reply, I was really busy during the last weeks. Could you try to install [this build](https://1drv.ms/u/s!ApwzDE01ByurhzVdsw4tvJrF08Jn) of the QDriverStation and report how it behaves?

Qormix, 2017-02-11:
Can't get the build to install. Missing dependencies qt58base, qt58declarative.

Qormix, 2017-02-11:
Anyways built if from src. Seems to be working. 
[10_53_17 AM.txt](https://github.com/FRC-Utilities/QDriverStation/files/768597/10_53_17.AM.txt)

snapp-development, 2017-02-11:
I'm also having trouble with missing dependencies. I attempted to build from source but on Debian I still have no communication.

alex-spataru, 2017-02-11:
Sorry, it seems that the build script called Qt 5.8's qmake instead of qt5-qmake, i'll fix that in a moment.

alex-spataru, 2017-02-12:
Here is the fixed Debian package: https://1drv.ms/u/s!ApwzDE01ByurhzYnQ75laNnvNVfG

alex-spataru, 2017-02-12:
@rolylolypoly  @Qormix  @snapp-development  Does the new version work consistently? Or do you still experience communication issues?

alex-spataru, 2017-02-14:
I think I found the issue, it has to do with how sockets where closed by the QDriverStation. 

Could anyone try this build (https://1drv.ms/u/s!ApwzDE01ByurhzgD8eRy_Nc5x3Yo) to check if that fixes the problem?

snapp-development, 2017-02-14:
@alex-spataru I will try to install it on a Debian machine at some point today to test.

alex-spataru, 2017-02-14:
@snapp-development I just found out that the build script added some Yakkety-specific depedencies. Please download this build instead (to avoid dependency issues): https://www.dropbox.com/s/4ocsq7f56dwoxbu/qdriverstation_17.02_generic.deb?dl=0

alex-spataru, 2017-04-16:
Any updates on this issue whatsoever? Since I don't have access to any FRC robots near my area (moved to another city for college), I rely on the information and feedback given by users of the QDS to fix comm. related issues.

calebeby, 2017-05-12:
@alex-spataru That build worked for me. Thanks!

calebeby, 2017-05-12:
Could you put that on the releases page so people install it rather than 17.01.1?

alex-spataru, 2017-05-13:
@calebeby Thanks for the information! I'll draft a new release soon!

alex-spataru, 2017-05-17:
I just released a new version of the QDriverStation, hopefully it fixes these issues.

BearzRobotics, 2017-05-17:
I'll try to get it packaged this weekend for solus.

scottm314, 2017-05-21:
I am having an issue with connecting the robot (No Robot Communications).  I am running OS X 10.12.4.  I can deploy code with eclipse and I can ping the roborio.  I have been following this chain for a while and waited for the 17.05 program before posting.  I am connected through ethernet.  The interesting thing is that I connected a windows PC using the official drive station through the USB connection (after failing to connect) and I get messages on the console.  I am connecting through DHCP.  After failing to connect with automatic I tried all of the likely IP addresses by setting it manually.  Any thoughts?  

log below.........

Log created on:      May 21 2017 - 14:20:18 PM
LibDS version:       2.2
Operating System:    OS X 10.12
Application name:    QDriverStation
Application version: 17.05

------------------------------------------------------------------------
ELAPSED TIME   ERROR LEVEL   MESSAGE     
------------------------------------------------------------------------
00:00.2        DEBUG         Scale factor set to: 1
00:00.7        DEBUG         DS Client: Changing team number to 6488
00:00.7        DEBUG         DS Events: Team number set to 6488
00:00.7        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:00.7        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:00.7        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:00.7        DEBUG         DS Client: Using new FMS address ""
00:00.7        DEBUG         DS Client: Using new radio address ""
00:00.7        DEBUG         DS Client: Using new robot address ""
00:00.7        DEBUG         DS Client: Switched to FRC 2016 Protocol
00:00.7        DEBUG         Dashboard command set to: java -jar "/Users/Scott/wpilib/tools/sfx.jar"
00:01.7        DEBUG         DS Client: Resetting joysticks...
00:01.7        DEBUG         DS Client: Using new FMS address ""
00:01.7        DEBUG         DS Client: Using new radio address ""
00:01.7        DEBUG         DS Client: Using new robot address ""
00:01.7        DEBUG         DS Client: Resetting joysticks...
00:01.7        WARNING       QProcess::start: Process is already running
00:01.7        DEBUG         Dashboard command set to: java -jar "/Users/Scott/wpilib/tools/sfx.jar"
00:01.7        DEBUG         DS Client: Setting position to 0
00:01.7        DEBUG         DS Client: Setting alliance to 0
00:01.7        DEBUG         Initialized in  1768 milliseconds
00:01.8        WARNING       QProcess::start: Process is already running
00:01.8        WARNING       QProcess::start: Process is already running
00:05.7        DEBUG         DS Events: FMS communications set to false
00:05.7        DEBUG         DS Events: Robot code status set to false
00:05.8        DEBUG         DS Events: Robot enabled state set to false
00:05.8        DEBUG         DS Events: ESTOP set to false
00:05.8        DEBUG         DS Events: Robot communications set to false
00:07.2        DEBUG         DS Client: Using new FMS address ""
00:07.2        DEBUG         DS Client: Using new radio address ""
00:07.2        DEBUG         DS Client: Using new robot address ""
07:47.1        DEBUG         DS Client: Setting enabled state to false
07:48.6        DEBUG         DS Client: Setting enabled state to false
07:56.4        DEBUG         DS Client: Using new FMS address ""
07:56.4        DEBUG         DS Client: Using new radio address ""
07:56.4        DEBUG         DS Client: Using new robot address ""
44:35.3        DEBUG         DS Client: Using new FMS address ""
44:35.3        DEBUG         DS Client: Using new radio address ""
44:35.3        DEBUG         DS Client: Using new robot address ""
53:26.2        DEBUG         DS Client: Using new FMS address ""
53:26.2        DEBUG         DS Client: Using new radio address ""
53:26.2        DEBUG         DS Client: Using new robot address ""
53:34.7        DEBUG         DS Client: Using new FMS address ""
53:34.7        DEBUG         DS Client: Using new radio address ""
53:34.7        DEBUG         DS Client: Using new robot address ""
53:50.9        DEBUG         DS Client: Using new FMS address ""
53:50.9        DEBUG         DS Client: Using new radio address ""
53:50.9        DEBUG         DS Client: Using new robot address ""

alex-spataru, 2017-05-22:
@scottm314 For some reason the robot address keeps changing to an empty or invalid IP, also, it seems that the robot address was set to 0.0.0.0 when launching the QDriverStation. Are you sure you clicked the Apply button when testing the likely robot IPs?

scottm314, 2017-05-23:
It is a good question, it would not be the first time I forgot to do something like that!  But, in this case, I am sure that I hit the apply button.  I did a ping after each change to make sure that I was connected t to the roborio.  

I did notice the 0.0.0.0 and was also interested to see that the program "Remembered" the team number between updates, even when I  deleted the entire old version directory.  I was thinking that there must be somewhere I could hardwire the ip address since there must be a data file that has the team number but I could not find it.

Tommsy64, 2017-08-25:
I am having a similar issue as @scottm314. Full log file [here](https://pastebin.com/9H4JZ8A5). (Running OSX 10.12.6, roboRIO 2017)
```
LibDS version:       2.2
Operating System:    macOS Sierra (10.12)
Application name:    QDriverStation
Application version: 17.05
```

I can ping the roboRIO through mDNS, SmartDashboard connects, and I can deploy robot code.

The robot address in the settings window appears to reset every time I open the client, even after hitting both **Apply** and **Ok**.

Tommsy64, 2017-08-25:
I just compiled the ConsoleDS with the roboRIO's IP and it connects without errors.

alex-spataru, 2017-08-26:
@Tommsy64 Judging from the log file, you set a custom IP for the radio, not the robot (lines 39 & 40). Could you please try to check your settings window and see if the custom IPs are set correctly?

Tommsy64, 2017-08-26:
Oops! Setting the IP for the *robot* works; however, I have to turn on QDriverStation on a separate WiFi network and then switch to the robot's network for it to establish comms.

Turning on the driver station while connected to the robot's network—even switching back and forth between other networks—does not work. [Log](https://pastebin.com/e123vMgp). Nevertheless, this will be sufficient.

alex-spataru, 2017-08-28:
@Tommsy64 That's good news! Hopefully the issue is not so complex...does the QDriverStation manage to connect directly to your robot now?

bornamm, 2017-08-31:
i installed the QDriverStation and for some reason i can't get any communication with the bot. i tried reading other users issues but i simply couldn't figure anything out. 

Log created on:      Aug 31 2017 - 14:16:27 PM
LibDS version:       2.2
Operating System:    OS X 10.12
Application name:    QDriverStation
Application version: 17.05

------------------------------------------------------------------------
ELAPSED TIME   ERROR LEVEL   MESSAGE     
------------------------------------------------------------------------
00:00.5        DEBUG         Scale factor set to: 1
00:04.3        DEBUG         DS Client: Changing team number to 3019
00:04.3        DEBUG         DS Events: Team number set to 3019
00:04.3        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:04.3        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:04.3        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:04.3        DEBUG         DS Client: Using new FMS address ""
00:04.3        DEBUG         DS Client: Using new radio address ""
00:04.3        DEBUG         DS Client: Using new robot address "roboRIO-3019.local"
00:04.3        DEBUG         DS Client: Switched to FRC 2015 Protocol
00:04.8        DEBUG         DS Client: Resetting joysticks...
00:04.8        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:04.8        DEBUG         DS Client: Using new FMS address ""
00:04.8        DEBUG         DS Client: Using new radio address ""
00:04.8        DEBUG         DS Client: Using new robot address "roboRIO-3019.local"
00:04.9        DEBUG         DS Client: Resetting joysticks...
00:04.9        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:04.9        DEBUG         DS Client: Setting position to 0
00:04.9        DEBUG         DS Client: Setting alliance to 0
00:04.9        DEBUG         Initialized in  4924 milliseconds
00:04.9        DEBUG         DS Client: Resetting joysticks...
00:04.9        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:05.4        WARNING       QProcess::start: Process is already running
00:05.4        WARNING       QProcess::start: Process is already running
00:05.4        WARNING       QProcess::start: Process is already running
00:09.3        DEBUG         DS Events: FMS communications set to false
00:09.3        DEBUG         DS Events: Robot code status set to false
00:09.3        DEBUG         DS Events: Robot enabled state set to false
00:09.3        DEBUG         DS Events: ESTOP set to false
00:09.3        DEBUG         DS Events: Robot communications set to false

alex-spataru, 2017-09-01:
@bornamm You are using the 2015 communication protocol, which is essentially the same as the 2016 protocol, but with a different robot address. Try switching to the 2016 protocol, if mDNS does its job, the QDriverStation should be able to communicate with your robot. If not, ping your robot from terminal to obtain the robot's IP, and input the obtained IP in the 'custom robot address' field in the settings dialog. 

Please let me know if QDS worked, or if you need further assistance.

bornamm, 2017-09-01:
What should i type in the mac terminal to ping it? (ping roboRIO-####-FRC.local where #### is the team number?) also what does this pinging do?

bornamm, 2017-09-01:
So I pinged it n got the IP address for the bot n changed the ip adress on the qdriver station, n it still got no robot communication with this log:

Log created on:      Sep 01 2017 - 14:07:20 PM
LibDS version:       2.2
Operating System:    OS X 10.12
Application name:    QDriverStation
Application version: 17.05

------------------------------------------------------------------------
ELAPSED TIME   ERROR LEVEL   MESSAGE     
------------------------------------------------------------------------
00:00.6        DEBUG         Scale factor set to: 1
00:04.2        DEBUG         DS Client: Changing team number to 3019
00:04.2        DEBUG         DS Events: Team number set to 3019
00:04.2        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:04.2        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:04.2        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:04.2        DEBUG         DS Client: Using new FMS address ""
00:04.2        DEBUG         DS Client: Using new radio address ""
00:04.2        DEBUG         DS Client: Using new robot address "169.254.40.67"
00:04.2        DEBUG         DS Client: Switched to FRC 2016 Protocol
00:04.7        DEBUG         DS Client: Resetting joysticks...
00:04.7        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:04.8        DEBUG         DS Client: Using new FMS address ""
00:04.8        DEBUG         DS Client: Using new radio address ""
00:04.8        DEBUG         DS Client: Using new robot address "169.254.40.67"
00:04.8        DEBUG         DS Client: Resetting joysticks...
00:04.8        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:04.8        DEBUG         DS Client: Setting position to 0
00:04.8        DEBUG         DS Client: Setting alliance to 0
00:04.8        DEBUG         Initialized in  4886 milliseconds
00:05.1        DEBUG         DS Client: Resetting joysticks...
00:05.1        DEBUG         DS Client: Registered new joystick with 6 axes, 0 hats and 15 buttons
00:05.3        DEBUG         DS Events: FMS communications set to false
00:05.3        DEBUG         DS Events: Robot code status set to false
00:05.3        DEBUG         DS Events: Robot enabled state set to false
00:05.3        DEBUG         DS Events: ESTOP set to false
00:05.3        DEBUG         DS Events: Robot communications set to false
00:49.1        DEBUG         DS Events: Robot enabled state set to false

alex-spataru, 2017-09-01:
@bornamm Thanks! (and sorry for not responding to your first message) We now know that mDNS is not to blame for your communication issues.

If possible, please:
- Verify that no one else is connected to the robot
- Verify that there are no other instances of the QDS running on your computer

From the logs that I received, it seems that most communication issues are happening under Mac OS Sierra. Can anyone with a Mac use Wireshark to check the packets that are sent/received between the robot and the QDriverStation? This could help us clarify what is causing this issue.

closed 2020-10-07 by alex-spataru