FRC-Utilities/QDriverStation #46: QDriverStation recognizes joysticks but does not receive inputs (MacOS only)

Masterzach32, 2017-01-17:
Title. We have tried two different computers (one on sierra and one on mavericks), a logitech attack 3 and logitech attack 3d.

Anyone else experiencing this issue?

alex-spataru, 2017-01-18:
Thanks for the report! Could you please upload your log file? (you can find them in ~./qdriverstation). Also, did you try to use any other joystick (e.g. an Xbox controller) with your computers?

Masterzach32, 2017-01-18:
Yes, I will be at our team's shop tonight, so I will do it then, but the Logitech Attack 3 and 3D are unfortunately the only joysticks we have. Ill ask someone to bring an xbox controller tonight.

alex-spataru, 2017-01-18:
I replicated your issue, I will upload a patch in a few moments. Thanks for the report!

alex-spataru, 2017-01-18:
I released another version which fixes your issue. You can update the QDriverStation just by opening it when you have an Internet connection or by downloading it from here: https://github.com/FRC-Utilities/QDriverStation/releases/latest

Just to be sure, could verify that the new version works for you too before closing this issue?

Masterzach32, 2017-01-18:
Works like a charm. Thank you!

closed 2017-01-18 by Masterzach32