FRC-Utilities/QDriverStation #47: smartdahsboard not working

quillford, 2017-01-26:
smartdashboard worked fine with the 2016 control system and the qdriverstation 16, but it's not working with the 2017 control system and qdriverstation 17

alex-spataru, 2017-01-27:
Hello, could you please describe what part of the SmartDashboard is not working? (e.g. if the QDriverStation does not open the SmartDashboard).

Thanks!

quillford, 2017-01-27:
it doesn't open. if i manually open it, nothing comes up.

alex-spataru, 2017-01-27:
Thanks, it is possible that the location of the SmartDashboard has changed. Just to be sure what is happening, could you upload one of your log files here?

quillford, 2017-01-27:
https://hastebin.com/kacozeboba.sql

quillford, 2017-01-27:
that's from 17.01.1

alex-spataru, 2017-01-27:
Judging from the log file, it seems that QDriverStation tries to open your dashboard and does nothing, which probably means that the dashboard location or command has changed. I'll look at this issue tomorrow morning. 

Thanks for the report!

closed 2019-09-17 by quillford