FRC-Utilities/QDriverStation #48: No z Axis

BearzRobotics, 2017-02-16:
On the Logitech joystick that came with the kit of parts a few years back, QDriverStation can't read the z value of the controller.

[Logs](https://github.com/FRC-Utilities/QDriverStation/files/778800/Wed.15.zip)

alex-spataru, 2017-02-18:
Hello, did you try to test the joystick with other applications (such as the official FRC Driver Station)?

BearzRobotics, 2017-02-18:
Yes, it works fine in the official driver station. It also didn't work in the [robotpy](http://robotpy.readthedocs.io/en/stable/) simulation which uses pygame (which uses sdl for joystick input.)

alex-spataru, 2017-02-19:
It is possible that this issue can be fixed by registering your joystick to the [database](https://github.com/FRC-Utilities/QDriverStation/blob/master/lib/QJoysticks/etc/resources/SDL/Database.txt) file. Could you tell me the exact model of your joystick please?

closed 2017-07-24 by BearzRobotics