FRC-Utilities/QDriverStation #52: Robot code not detected

OttoDeLupe, 2017-10-29:
I'm running 17.05, downloaded as a .deb on Ubuntu 16.04.

When starting, comms are established with the rio, but I get the following in the logs:
00:00.6        DEBUG         DS Events: Robot code status set to false

I'm able to deploy the java code from eclipse to the rio just fine. So, am not sure what might be going on. Also, as far as I can see, if my code what throwing an exception, there's no log that tells me that - or maybe it's just not getting that far.

Also - since I'm connected to the rio via USB networking (IP address 172.22.11.2), I'm unable to edit the network addresses from qdriverstation. I worked around this by putting an entry in /etc/hosts that resolves roboRIO-xxxx-FRC.local to 172.22.11.2. Not sure if this is a known issue or not.

alex-spataru, 2017-10-29:
Thanks for the feedback, it may be possible that you have selected the wrong DS protocol version from the settings tab. To clarify the cause of your issue, could you please upload the full log? (You can find them in ~./qdriverstation, or by clicking the "Logs..." button in the console section).

OttoDeLupe, 2017-10-29:
Log file as requested:

Log created on:      Oct 29 2017 - 13:26:07 PM
LibDS version:       2.2
Operating System:    Ubuntu 16.04.3 LTS
Application name:    QDriverStation
Application version: 17.05

------------------------------------------------------------------------
ELAPSED TIME   ERROR LEVEL   MESSAGE     
------------------------------------------------------------------------
00:00.3        DEBUG         Scale factor set to: 1
00:00.5        DEBUG         DS Client: Changing team number to 1289
00:00.5        DEBUG         DS Events: Team number set to 1289
00:00.5        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:00.5        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:00.5        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:00.5        DEBUG         DS Client: Using new FMS address ""
00:00.5        DEBUG         DS Client: Using new radio address ""
00:00.5        DEBUG         DS Client: Using new robot address ""
00:00.5        DEBUG         DS Client: Switched to FRC 2016 Protocol
00:00.5        DEBUG         DS Client: Resetting joysticks...
00:00.5        DEBUG         DS Client: Using new FMS address ""
00:00.5        DEBUG         DS Client: Using new radio address ""
00:00.5        DEBUG         DS Client: Using new robot address ""
00:00.5        DEBUG         DS Client: Resetting joysticks...
00:00.5        DEBUG         DS Client: Setting position to 0
00:00.5        DEBUG         DS Client: Setting alliance to 0
00:00.5        DEBUG         Initialized in  585 milliseconds
00:00.6        DEBUG         DS Events: Robot code status set to false
00:00.6        DEBUG         DS Events: ESTOP set to false
00:00.6        DEBUG         DS Events: Robot communications set to true
00:01.5        DEBUG         DS Events: FMS communications set to false

alex-spataru, 2017-11-01:
Thanks! Unfortunately it seems that the issue has something to do with QDriverStation, I will check the protocol code later this week. I have to finish some freelance work these days.

OttoDeLupe, 2017-11-01:
Thank you - sorry for the delay in responding. - I've been dealing with the effects of a wind storm that took out electricity for a few days.

Further data point - with 2016 protocol, I do get robot comms. With 2015 and 2014, there is no robot comms. You probably know that (given the log file), but just wanted to let you know.

OttoDeLupe, 2017-11-03:
Found myself at work today with some extra time, so I took a dive in. Here's a long shot guess:

frc_2015.c line 71, uint8_t cRobotHasCode = 0x20
Since cRobotHasCode is a uint8_t, this will truncate the supplied 16bit value, resulting in cRobotHasCode getting set to zero. 

In read_robot_packet(), line 642, this zero value gets AND'd  to the status bits, resulting in the robot's code status getting set to zero.

I'm guessing this was a typo and cRobotHasCode should be set to 0x02 

Since I installed from a .deb; I'll try to set up an actual build/test environment this weekend & see if I'm right.

alex-spataru, 2017-11-04:
Thanks for the information, I will check again the 2015 protocol implementation (I did not check that code for almost a year :laughing:).

closed 2020-10-07 by alex-spataru