FRC-Utilities/QDriverStation #54: Installation fails on Ubuntu 17.10

JackToaster, 2017-12-04:
I tried to install QDriverstation 17.05, but the installation fails with the following output:
```Reading package lists... Done
Building dependency tree        
Reading state information... Done
Reading state information... Done
This package is uninstallable
Dependency is not satisfiable: libqt5declarative5 (>= 5.0)

```
I have also tried installing older versions, but the installation fails in the same way with every version I've tried.

BearzRobotics, 2017-12-04:
I presume you tried to install the pre built Deb? If so that Deb file will only work on the version of Ubuntu it was built on. The reason for this is because during the installation it won't be able to find the correct version of dependency . The problem with just letting it install with out the correct version is that the abi of the library could of changed. If this is the case you'll have to compile from source.

JackToaster, 2017-12-09:
Oh, that makes sense. Compiling from source works just fine for me.

closed 2017-12-09 by JackToaster

BearzRobotics, 2017-12-09:
I was just about to come and see if you were able to get it working. I'm glade your were able to figure it out.