FRC-Utilities/QDriverStation #55: runtime  error in linux

guoxiangyang, 2018-01-10:
after succefull make with no error, run the qdriverstation got the error below:

exec from local folder or after sudo make install got the same message
```
gxy@local:~/work/frc/QDriverStation$ ./qdriverstation 
00:00.4        WARNING       QQmlApplicationEngine failed to load component
00:00.4        WARNING       qrc:/qml/main.qml:72 Type MainWindow unavailable
qrc:/qml/MainWindow/MainWindow.qml:27 module "Qt.labs.settings" is not installed

00:00.4        WARNING       QProcess: Destroyed while process ("bash") is still running.
00:00.4        WARNING       QProcess: Destroyed while process ("bash") is still running.
```

```
gxy@local:~/work/frc/QDriverStation$ lsb_release -a
LSB Version:    core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID: LinuxMint
Description:    Linux Mint 18.2 Sonya
Release:        18.2
Codename:       sonya
```

```
gxy@local:~/work/frc/QDriverStation$ qmake -v
QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu
```

```
gxy@local:~/work/frc/QDriverStation$ uname -a
Linux local 4.10.0-40-generic #44~16.04.1-Ubuntu SMP Thu Nov 9 15:37:44 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

alex-spataru, 2018-01-16:
Your issue is caused because you don't have the QSettings QML module installed, try installing [this](https://packages.ubuntu.com/xenial/libs/qml-module-qt-labs-settings) package or downloading the latest version of the QDriverStation (in AppImage format).

Sorry for the late reply!

closed 2018-01-16 by alex-spataru

guoxiangyang, 2018-01-16:
It works!

Thank you very much.