FRC-Utilities/QDriverStation #56: Game Specific Message

codeNinjaDev, 2018-01-21:
Is there the game data for FIRST Power Up, such as the location of the switches and scale?

BearzRobotics, 2018-01-21:
There might be, but such info doesn't come from the driverstation. The best place to look would be the  [game manual](https://www.firstinspires.org/resource-library/frc/competition-manual-qa-system). Section 3 Arcade has a break down of the field. Section 3.9 has vision targets that might also be of help.

auscompgeek, 2018-02-08:
> There might be, but such info doesn't come from the driverstation.

[It does.](http://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/826278-2018-game-data-details)

BearzRobotics, 2018-02-09:
Okay thanks for the correction. This is not how it was done, in the past. I only gave it a glance right now, but I'll read it through this weekend.

alex-spataru, 2020-10-07:
Closing this issue since we'll be using a new DS library.

closed 2020-10-07 by alex-spataru