FRC-Utilities/QDriverStation #57: QDriverStation detects joysticks but doesn't send to robot

retrodaredevil, 2018-01-25:
Information on what I'm running:
Log created on:      Jan 24 2018 - 18:34:53 PM
LibDS version:       2.2
Operating System:    Ubuntu 16.04.3 LTS
Application name:    QDriverStation
Application version: 18.01

The problem is the same as this: https://github.com/FRC-Utilities/QDriverStation/issues/46 except on linux. The log connects, enables, disable the robot fine and the SmartDashboard even works perfectly except when I try to use a joystick on it, it detects it, but doesn't send it and I get an error when I call a joystick.getButton() in my code.

The log doesn't have any errors in it, just debug although I'd be happy to add other log files if that would help.

Another thing that would be nice is to be able to see the ports that the joysticks are on since right now it just shows a list of joysticks. I've tried this with a ps4 controller and a generic joystick. I've only tested this on my computer but I have yet to get the controllers working.

alex-spataru, 2018-01-25:
Could you try to use the virtual joystick to control your robot (and report back the results)? This can help me to know if this is an issue with SDL, or with the LibDS library itself.

Thanks!

retrodaredevil, 2018-01-27:
Tested it with virtual joystick and nothing.

fearxzombie, 2018-01-31:
Same problem here, on Elementary OS Loki.

alex-spataru, 2018-02-04:
Sorry for the late reply. Maybe the communication protocol has changed a little bit since last year, hopefully this is not the case.

fearxzombie, 2018-02-04:
I think it probably is the case, because the robots I tested it with were 2018 robots
		
		

		-Gabriel McMillan

alex-spataru, 2018-02-04:
Does your team still have access to 2015-2017 robots? If so, could you please try controling those robots with the QDriverStation?

Unfortunately, I am not into FRC anymore due to college, which makes testing LibDS and QDriverStation very difficult for me.

fearxzombie, 2018-02-04:
I do, however it will be a while before I can test it, I will get back to you on that.
		
		

		-Gabriel McMillan

eamonniknafs, 2018-02-04:
I have a 2016 robot and a current one. I’ll check on that on Monday, if no one has by then.

fearxzombie, 2018-02-04:
Got it.
		
		

		-Gabriel McMillan

alex-spataru, 2018-08-10:
We'll move discussion of this issue to #58, which deals with the 2018 protocol implementation. Thanks for the reports!

closed 2018-08-10 by alex-spataru