FRC-Utilities/QDriverStation #58: 2018 Protocol?

eamonniknafs, 2018-01-28:
Will QDriverStation support the 2018 protocol, or has this been abandoned?

BearzRobotics, 2018-01-29:
I depends. I don't have access to a robot and I believe Alex doesn't either, since we're both graduated and off to college. So the project is really in maintenance mode.

eamonniknafs, 2018-01-29:
Ah, got it. Do you know where I can find documentation about the new protocol? I can probably figure it out and make a pull request when I do.

riftware, 2018-01-29:
I'm very interested too...I'm a Java developer that hasn't done much c or
c++ in a while but I might try to take a look at it

BearzRobotics, 2018-01-29:
I'm glad that there is some interest. As for as the 2018 protocol the only thing I could find was [this](https://wpilib.screenstepslive.com/s/currentCS/m/troubleshooting/l/705152-fms-whitepaper). It's for the FMS which uses different ports than when hooked up directly. Other than that, the protocols were reversed engineered using Wireshark by projects before this one. The actual code that communicates with the bot is in this other [repo](https://github.com/FRC-Utilities/LibDS). If it is like the last two years, then its simple changes to the already established protocol. For example, the 2016/2017 protocol only changed the default name of the robot for mdns lookup, from the 2015 protocol.

HeroCC, 2018-01-31:
I could look into preforming a few wireshark dumps from our NI driver station if it would help! 2018 support would be great

BearzRobotics, 2018-01-31:
Yee, it would. Anything I do though will have to be tested by someone else.

StripedMonkey, 2018-04-09:
Is there any progress on this? I'd like to try to work on this if possible but I find networking stuff a daunting task.

jacksondm33, 2018-04-09:
I have done some wireshark dumps recently to try to make the netconsole work.
Other than that and a few other minor problems, everything seems to work in 2018.
Also, most/all the networking is done through LibDS which QDriverStation uses.

jacksondm33, 2018-04-09:
I found that FRC Driver Station is communicating with the robot netconsole over tcp port 1740. I'm not sure if it still does udp port 6666 dumps. If not, I'm not sure when this was changed, but it makes sense to use tcp.

alex-spataru, 2018-04-11:
Unfortunately, that's the best documentation that I have found. FIRST and NI do not provide any documentation on the FRC Communication protocols. So essentially, we need to reverse engineer with Wireshark any new communication protocol for the moment.

jcreigh, 2018-04-12:
I've started to rewrite my dissectors and update my documentation, but I've _reeeeally_ lazy. I guess if people are interested I can start working on it slightly faster. And if anyone has specific questions (like this one) I can try to see what I can find out, ping me or just email me or something. I have a lot of competition packets to wade through, and I'll probably collect some more... I guess starting tomorrow.

And I can confirm, stdout is now being sent through the tcp channel on port 1740 and not udp port 6666. The [encapsulation](https://github.com/jcreigh/FRCDriverStation/wiki/DS-%E2%86%94-RoboRIO-TCP) is the same, but there's a couple new packet types this year. The main ones are:

Src | ID | Packet Type
---|---|---
RIO | `0x0c` | stdout
DS | `0x0e` | Game data 

Looks like each stdout message is prefixed with 6 bytes of stuff. Haven't looked into it.

jcreigh, 2018-04-12:
Alright, I've updated the documentation for the TCP channel. Didn't try to figure out the data at the beginning of standard output and error log packets yet. I'll make a dissector when I'm feeling motivated to, unless someone wants to grease my wheels. ;p

[DS ↔ RoboRIO TCP](https://github.com/jcreigh/FRCDriverStation/wiki/DS-%E2%86%94-RoboRIO-TCP)

jacksondm33, 2018-04-14:
Thanks @jcreigh . This documentation is very useful as I don't have a lot of wireshark data. How do you get so many competition packets? Are you actually running wireshark on the DS computer during matches?

jcreigh, 2018-04-14:
tshark and a batch script to set it up, yep.

jcreigh, 2018-04-22:
Alright, status update. I still need to work on the wireshark dissectors, but I've updated the documentation to 2018 and added pretty much all the rest of the information that was missing. There's a couple little things, but nothing major.

https://github.com/jcreigh/FRCDriverStation/wiki

Have fun.

StripedMonkey, 2018-04-24:
Cool, I hope I can actually read some of this and understand it :)

StripedMonkey, 2018-08-09:
Unfortunately no, I've been looking at it and had a few false starts, but its overall been negligible in terms of progress. I could give it a go but the problem lies in the fact that I don't have access to a RIO over the summer so any changes I make are "I think this will fix the issue" but not actually proven.

StripedMonkey, 2018-08-10:
I'll contact you and we can try and work something out! Its good to see someone willing to help out some :)

jordanbancino, 2018-08-10:
@StripedMonkey Awesome! I'll wait for you to contact me. If you'd like, my email is available on my profile.

@alex-spataru I will be testing on a robot-type board. So it's basically a PDP, RoboRIO, radio, and a few motors with Talons zip tied to a board. It won't be going anywhere :laughing:. 
Also, thanks for your continued support. I use Linux on my laptop, so in order to use the Driver Station, I have Windows 7 installed in a VM. This project would be helpful because it would eliminate my need for a VM altogether.

I have a quick question though. Will this driver station eventually support pushing game data to a robot? I noticed that when I used the latest release on the releases page, there was no place to input game data. This isn't a requirement of mine, just something I'm wondering. Maybe I should open an issue for it? I'm not sure how any of this works, so I have no idea how easy or difficult it would be to implement.

jordanbancino, 2018-08-10:
It appears that #56 relates to my question about game data, which I think is delivered from the FMS if connected, but I know the FRC DS has an option to input your own, if you aren't connected to the FMS.

alex-spataru, 2018-08-22:
> I have a quick question though. Will this driver station eventually support pushing game data to a robot? I noticed that when I used the latest release on the releases page, there was no place to [...]

Currently, LibDS and the QDriverStation do not support FRC Game Data, I can create some dummy functions for LibDS so that they can be implemented (maybe an `ExtraNetworking` section or something similar). I'll download a copy of the latest FRC Driver Station this weekend so that I can begin working on the UI part. I will post updates about this in issue #56.

alex-spataru, 2019-01-03:
As of the latest commit, I have added [frc_2018.c](https://github.com/FRC-Utilities/LibDS/blob/master/src/protocols/frc_2018.c), which will serve as a base for implementing the 2018 protocol. The file contains some tips and instructions to help anyone who is interested to continue development of this project. 

I have also updated the UI-related functions so that the FRC 2018 protocol can be selected and loaded by the user directly from the QDriverStation (or any spin-off project).

BTW, happy new year!

Redrield, 2019-01-03:
In the mean time for anyone using Linux who wants a DS with 2018 protocol support, I've written [consoleds](https://gitlab.com/Redrield/consoleds) based on the documentation from jcreigh. Supports game data and joysticks. I haven't gone so far as to include FMS and such, seeing as it will never be comp legal anyhow.

alex-spataru, 2019-01-03:
@Redrield Thanks! I'll take a look at your project to see if I can begin working on the 2018 protocol before I start my quarter at college. I'll keep you updated through this thread! I just hope that FIRST didn't change the comm. protocols for 2019.

jacksondm33, 2019-02-24:
I don't think too much has changed in the 2018/2019 protocols as QDriverStation works fine with them as far as I have seen. One thing that doesn't work is the netconsole (used for messages and stdout) which I was recently able to fix by adding tcp sockets and fixing a flaw in their creation. I now can get info and prints from what I have tested. I have not added complete support for reading tcp packets nor have I implemented sending tcp packets, though. Also, my changes seems to might have caused some crashes to occur sometimes (double frees, segfaults). If I have time I will continue working on it, but right now basic functionality seems to be working. See [my fork](https://github.com/jacksondm33/LibDS/tree/feature/netconsole) for my changes.

jordanbancino, 2019-02-25:
Oh sweet! This project just came to mind the other day, I was just wondering if any progress was being made. Then I saw this. I'll definitely take a look at your fork, because I am very interested in using LibDS with ConsoleDS this year for testing.

Redrield, 2019-02-26:
Not sure if ConsoleDS is in reference to my project, but it doesn't use LibDS, rather uses a rust implementation of the DS protocol available on crates.io

jordanbancino, 2019-02-26:
@Redrield I am referring to the sample project “ConsoleDS” in the LibDS examples folder. 

https://github.com/jacksondm33/LibDS/tree/feature/netconsole/examples/ConsoleDS

Redrield, 2019-02-26:
Gotcha, I should really come up with a more creative name for my project

jordanbancino, 2019-02-27:
@Redrield Haha yeah, I always struggle with naming things. Probably the hardest part of programming 🤣

alex-spataru, 2020-10-07:
Closing this issue since it's no longer relevant. Check #62 for more info.

closed 2020-10-07 by alex-spataru