FRC-Utilities/QDriverStation #59: Submodules

jacksondm33, 2018-04-10:
QDriverStation relies on LibDS and QJoysticks which are two separate git repositories. Instead of just copying the code whenever there is an update, it would be better to make these submodules.

alex-spataru, 2018-04-11:
I think that's a good idea. I will make changes to the repository configuration and build instructions later this evening.

StripedMonkey, 2018-04-16:
Is this going to happen as a nested git repository?

StripedMonkey, 2018-06-24:
@alex-spataru *poke* I've started working on trying to update LibDS and it would be great if It wasn't nessicairy to update it in two places :) I can do it for you if you want.

alex-spataru, 2020-10-07:
Closing this issue since we now manage dependencies with submodules.

closed 2020-10-07 by alex-spataru