FRC-Utilities/QDriverStation #61: Fail to compile with fresh clone

patrickelectric, 2018-07-20:
```
../src/main.cpp:40:10: fatal error: EventLogger.h: No such file or directory
 #include <EventLogger.h>
```

alex-spataru, 2018-07-26:
Hi, sorry for the late reply!
I will try to compile the QDriverStation on my machine with a fresh clone. For the moment, could you please let me know your operating system and version?

alex-spataru, 2018-07-26:
As a side note, the LibDS is now included as a submodule, you may need to run the following commands from the repository root folder:
```
git submodule init
git submodule update
```
Alternatively, clone the repository again with the following command
`git clone --recursive https://github.com/FRC-Utilities/QDriverStation`

closed 2018-08-10 by alex-spataru

patrickelectric, 2018-08-10:
everything is working fine now, ty.