FRC-Utilities/QDriverStation #63: Please update for FRC 2020 protocol

iqv29, 2021-01-27:
Please update for FRC 2020 protocol

solonovamax, 2021-01-27:
project seems dead. There are some people who mentioned making a fork of it in #62, so I'd look at that.

alex-spataru, 2021-01-31:
@Boomaa23 implemented the 2020 comm. protocol (code [here](https://github.com/FRC-Utilities/LibDS/blob/ca68b52a9e58b758becedbfeab72c4507f23ec92/src/protocols/frc_2020.c), but its still not throughly tested. 

Since I haven't participated in FRC since 2017, I cannot work on the project without compromising the safety of anyone who uses the QDriverStation, since I cannot guarantee that the software behaves as expected. 

Given those reasons, I should not provide a binary build for users until the FRC 2020 protocol is tested and validated by its authors. If you have access to your robot and are willing to take the risks, you are welcome to [compile](https://github.com/FRC-Utilities/QDriverStation#build-instructions) & test the latest commit.

closed 2021-01-31 by alex-spataru