FRC-Utilities/QDriverStation #65: Weird garbled text encoding issue

FwappyPotato, 2021-09-24:
Running Arch with KDE Plasma, installed via AUR package. I presume it has something to do with character encoding schemes, but its strange its only happening with some text.

[image]

alex-spataru, 2021-09-24:
Hi! Can you please run QDriverStation from the terminal and paste its output here? I want to see if Qt notifies anything unusual regarding the text encoding.

FwappyPotato, 2021-09-25:
`00:00.1        DEBUG         Scale factor set to: 2.65567
00:00.3        WARNING       qrc:/qml/Dialogs/SettingsWindow.qml:98:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
00:00.4        WARNING       qrc:/qml/MainWindow/About.qml:35:5: QML Icon: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/About.qml:45:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/About.qml:55:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/About.qml:70:5: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/Messages.qml:36:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:156:9: QML Icon: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:162:9: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:169:9: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:175:9: QML Button: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:121:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
00:00.4        WARNING       qrc:/qml/MainWindow/Joysticks.qml:91:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
00:00.4        DEBUG         DS Client: Changing team number to 5419
00:00.4        DEBUG         DS Events: Team number set to 5419
00:00.4        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new FMS address ""
00:00.4        DEBUG         DS Client: Using new radio address ""
00:00.4        DEBUG         DS Client: Using new robot address ""
00:00.4        DEBUG         DS Client: Switched to FRC 2016 Protocol
00:00.4        WARNING       qrc:/qml/MainWindow/LeftTab.qml:133:9: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
00:00.4        DEBUG         DS Client: Resetting joysticks...
00:00.4        DEBUG         DS Client: Using new FMS address ""
00:00.4        DEBUG         DS Client: Using new radio address ""
00:00.4        DEBUG         DS Client: Using new robot address ""
00:00.4        DEBUG         DS Client: Resetting joysticks...
00:00.4        DEBUG         DS Client: Setting position to 0
00:00.4        DEBUG         DS Client: Setting alliance to 0
00:00.4        DEBUG         Initialized in  485 milliseconds
00:01.4        DEBUG         DS Events: FMS communications set to false
00:01.4        DEBUG         DS Events: Robot code status set to false
00:01.4        DEBUG         DS Events: Robot enabled state set to false
00:01.4        DEBUG         DS Events: ESTOP set to false
00:01.4        DEBUG         DS Events: Robot communications set to false
`
Thanks for taking a look!

FwappyPotato, 2021-10-09:
No change :c 
[image]

alex-spataru, 2021-10-09:
Damn, I truly have no idea what is causing this issue. I'll try to run the QDS on KDE Plasma on a virtual machine to investigate more on this issue.

Just please take into account that I am currently working & studying, it might take some time until I solve this out.

FwappyPotato, 2021-10-10:
No rush, I really appreciate the help!

I am using Intel Integrated graphics: 
[image]

bob8677, 2021-11-24:
I also had this problem, fixed it by installing the package `ttf-ubuntu-font-family` on Arch. Not sure why this happens as the program seems to include its own fonts, but it worked for me.

FwappyPotato, 2021-11-24:
Incredible!

[image]

alex-spataru, 2021-11-24:
Thanks a lot for the fix! I still don't know why this happens, in theory the font family is included within the application resources, and the AppImage bundle should handle every dependency.

AlexIsCool321, 2025-11-05:
I opened a pull request fixing this. I haven't used QT a lot but the font loading was wrong.
https://github.com/FRC-Utilities/QDriverStation/pull/72

BOBtistic, 2026-10-15:
I am a user on Ubuntu and I did use a LLM but I think i should post it for a permanent solution. I've only used Gear lever on this but this is what I did and it worked. I first went to terminal and typed in nano ~/.local/share/applications/qdriverstation.desktop (yours may be different). then I went down to Exec and imputed: Exec=env QT_QPA_PLATFORM=xcb QT_QUICK_BACKEND=software /home/USER/AppImages/qdriverstation.appimage (replace USER with your username). Then run: update-desktop-database ~/.local/share/applications/  . I don't know if this works for everyone but it worked for me on Ubuntu 26.04 GNOME. If it does not work, I am sorry. It worked for me so maybe it will work for you too.