FRC-Utilities/QDriverStation #67: AUR Package out of date

KenwoodFox, 2022-02-09:
The AUR package is way out of date: https://aur.archlinux.org/packages/qdriverstation
The appimage works though! https://aur.archlinux.org/packages/qdriverstation-appimage

If QDriverStation maintainers dont already have access to these packages, maybe that should happemn so they stay in sync?

Also as a note, excited to get my hands on a steam deck soon hopefully, ill be working to get steam deck inputs functioning for QDriverStation i hope!

lopsided98, 2022-03-16:
I am the maintainer of the AUR package. I don't use QDriverStation anymore, but I updated it to 21.04. If someone else wants to take it over, I'll give it to you.

KenwoodFox, 2022-03-17:
> I am the maintainer of the AUR package. I don't use QDriverStation anymore, but I updated it to 21.04. If someone else wants to take it over, I'll give it to you.

That would be really awesome, I think i could take over maintaining it if nobody else would like to instead