FRC-Utilities/QDriverStation #70: Crashes after changing a few values in driver station

Silicasandwhich, 2022-09-21:
Whenever I open driver station, it crashes a few moments later. Running in the terminal gives the following output (I changed team number and alliance position so I could tell when the program stops responding):

**Log:**
```
00:00.2        DEBUG         Scale factor set to: 1
00:00.4        DEBUG         DS Client: Using new FMS address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new radio address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new robot address "0.0.0.0"
00:00.4        DEBUG         DS Client: Using new FMS address ""
00:00.4        DEBUG         DS Client: Using new radio address ""
00:00.4        DEBUG         DS Client: Using new robot address ""
00:00.4        DEBUG         DS Client: Switched to FRC 2020 Protocol
00:00.4        WARNING       qrc:/qml/Widgets/Scrollbar.qml:130:5: QML Rectangle: Binding loop detected for property "height"
00:00.4        DEBUG         DS Client: Resetting joysticks...
00:00.4        DEBUG         DS Client: Registered new joystick with 6 axes, 1 hats and 10 buttons
00:00.4        DEBUG         DS Client: Using new FMS address ""
00:00.4        DEBUG         DS Client: Using new radio address ""
00:00.4        DEBUG         DS Client: Using new robot address ""
00:00.4        DEBUG         DS Client: Resetting joysticks...
00:00.4        DEBUG         DS Client: Registered new joystick with 6 axes, 1 hats and 10 buttons
00:00.5        DEBUG         DS Client: Setting position to 0
00:00.5        DEBUG         DS Client: Setting alliance to 0
00:00.5        DEBUG         Initialized in  513 milliseconds
00:01.3        DEBUG         DS Events: FMS communications set to false
00:01.3        DEBUG         DS Events: Robot code status set to false
00:01.3        DEBUG         DS Events: Robot enabled state set to false
00:01.3        DEBUG         DS Events: ESTOP set to false
00:01.3        DEBUG         DS Events: Robot communications set to false
00:03.4        DEBUG         DS Client: Setting position to 0
00:03.4        DEBUG         DS Client: Setting alliance to 1
00:03.4        DEBUG         DS Events: Team alliance set to 1
00:03.4        DEBUG         DS Events: Team position set to 0
00:04.8        DEBUG         DS Client: Setting position to 2
00:04.8        DEBUG         DS Client: Setting alliance to 0
00:04.8        DEBUG         DS Events: Team alliance set to 0
00:04.8        DEBUG         DS Events: Team position set to 2
00:04.8        DEBUG         DS Events: Team alliance set to 0
00:04.8        DEBUG         DS Events: Team position set to 2
00:05.9        DEBUG         DS Client: Setting position to 1
00:05.9        DEBUG         DS Client: Setting alliance to 1
00:06.0        DEBUG         DS Events: Team alliance set to 1
00:06.0        DEBUG         DS Events: Team position set to 1
00:06.0        DEBUG         DS Events: Team alliance set to 1
00:06.0        DEBUG         DS Events: Team position set to 1
00:08.3        DEBUG         DS Client: Changing team number to 1
00:08.3        DEBUG         DS Events: Team number set to 1
00:09.9        DEBUG         DS Client: Changing team number to 3
00:09.9        DEBUG         DS Events: Team number set to 3
00:10.2        DEBUG         DS Client: Changing team number to 34
00:10.2        DEBUG         DS Events: Team number set to 34
00:10.4        DEBUG         DS Client: Changing team number to 348
00:10.4        DEBUG         DS Events: Team number set to 348
00:10.5        DEBUG         DS Client: Changing team number to 3487
00:10.5        DEBUG         DS Events: Team number set to 3487
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
[1]    10504 IOT instruction (core dumped)  qdriverstation
```
Here is my neofetch in case any of the information would help.

[image]

 Honestly, in any other scenario I'd try and find a workaround myself, but this is too deep into c++ land for me. I’m struggling to understand what FD_SETSIZE even is lol

OnlyCS, 2022-11-15:
same issue here, also running Arch... Maybe this is an Arch related issue?

Silicasandwhich, 2022-11-15:
it's not arch exclusive, Ubuntu also has the problem. might be a gnome problem, I might test on kde.

AlokSwaminathan, 2023-02-16:
Same issue on Fedora KDE

Silicasandwhich, 2023-02-17:
so i kind of found a workaround for this problem by manually compiling the app natively for my system instead of using the appimage. however, it's kind of moot because i think FRC has changed the protocol for handling inputs, and i can only enable and disable my 2023 robot from QDriverStation, it can't send the robot joystick inputs. that part is way, *way* beyond my level of expertise.

Froggo8311, 2023-03-17:
Same issue on Arch, not GNOME, with no robot connected.

AdarWa, 2025-07-08:
Still having this issue with Ubuntu 24 LTS... Any fixes?

Art3mis2980, 2025-12-05:
same problem on Arch running plasma KDE, 

*** bit out of range 0 - FD_SETSIZE on fd_set ***: terminated
[1]    30998 IOT instruction (core dumped)  qdriverstation