FRC-Utilities/QDriverStation #71: Unable to connect to our robot on Mac

drckar, 2023-01-28:
our robot communication is done through wifi, when our mac is connected to the robot through wifi, QDriverStation says that there's no communication.

darthnithin, 2023-01-31:
Not sure if this project is still being updated or supported

drckar, 2023-02-04:
> 

it should still work though right?

alex-spataru, 2023-02-06:
Hi @drckar and @darthnithin! Unfortunately, I haven´t had access to FRC robots or components since 2016 (I moved to another city while in college). The QDriverStation should work with 2014, 2015, and 2016 robots. However, I cannot guarantee that it works with newer versions :(

If you are interested, you might want to contribute to [LibDS](https://github.com/FRC-Utilities/LibDS), a C library that I developed back in 2016 that handles all the Driver Station <-> Robot communications (QDriverStation is basically the "frontend", while LibDS is the "backend" of the project).

darthnithin, 2023-02-06:
I’ll take a look at LibDS but also yea it doesn’t work out of the box

blaze-developer, 2025-02-11:
> Hi [@drckar](https://github.com/drckar) and [@darthnithin](https://github.com/darthnithin)! Unfortunately, I haven´t had access to FRC robots or components since 2016 (I moved to another city while [...]

This looks very interesting. I am interested in doing something similar or contributing to get these projects working again. But I am not sure where to start reverse engineering the DS protocol. Do you have any resources or tips I could use while trying to update this project?

(P.S. I am making no promises of getting it working, this simply piqued my interest)

alex-spataru, 2025-02-11:
Hi @blaze-developer,

Reverse engineering the communication protocol can be done relatively easily with the right tools and a methodical approach. Here’s what you’ll need:

- Wireshark (https://www.wireshark.org)
- A computer with the official FRC Driver Station installed
- Patience, time, and a pen & paper for note-taking

### Basic Process:

1.	Install Wireshark on the computer running the FRC Driver Station.
2.	Connect to the robot’s WiFi access point.
3.	Start Wireshark and begin logging network traffic.
4.	Launch the FRC Driver Station and observe all interactions between your computer and the robot. I recommend taking notes on the robot’s state and the packets being sent.
5.	Introduce small changes (e.g., switching from teleoperated to autonomous mode) and log the differences.
6.	Repeat steps 3–5 as needed until you fully understand the data being transmitted under different conditions.
7.	Once you have a solid understanding, you can attempt to write your own driver station or modify LibDS — just be sure to take safety precautions.

### Additional Advice:

If you’re interested in developing a custom Driver Station, LibDS is a great reference and avoids the pain of reinventing the wheel. It’s written in C, a language I highly recommend learning (even after 12+ years in the programming world). 

While some people may find C challenging or "old", it provides great flexibility and a strong foundation for other languages like Python, Go, Java, ADA, or even FORTRAN. 

To get the most out of this process, I also suggest familiarizing yourself with hexadecimal notation, bytes, binary flags, etc — these concepts will be very useful when analyzing any kind of hardware/computer communication protocols.

Good luck, and most importantly, enjoy high school and college!